pbkhrv/rtl_433-hass-addons #1: Error loading shared library librtlsdr.so.0

nickF007, 2020-12-30:
Hi,

I installed the addon and i get this error:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf
Error loading shared library librtlsdr.so.0: No such file or directory (needed by /usr/local/bin/rtl_433)
Error relocating /usr/local/bin/rtl_433: rtlsdr_get_device_count: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_reset_buffer: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_get_tuner_gains: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_get_device_usb_strings: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_set_tuner_gain_mode: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_close: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_get_device_name: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_set_tuner_gain: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_get_sample_rate: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_read_async: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_set_center_freq: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_open: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_set_sample_rate: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_get_center_freq: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_get_index_by_serial: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_cancel_async: symbol not found
Error relocating /usr/local/bin/rtl_433: rtlsdr_set_freq_correction: symbol not found
[cmd] /run.sh exited 127
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.

```

My configfile looks like this:
```
output      mqtt://homeassistant.local:1883,user=******,pass=*******   <<I masked the usr en paswd

frequency   433.92M
protocol    15

convert     si
report_meta newmodel

```

As instructed i saved the file in /config/rtl_433/rtl_433.conf

Any idee how to fix this ? "ore what dit wrong"

Thanks.

pbkhrv, 2020-12-31:
Hi! Thanks for reaching out, sorry that it didn't work out of the box. Mind answering some questions so that I could help diagnose the issue?

What hardware are you running this on?
What version of Home Assistant OS are you using?

nickF007, 2020-12-31:
Hi Peter!

Thanks for your reaction. Of course, i will help with the diagnose! 

I also installed RTL433mqqt i had the same problem and solved it by adding:

```
export LD_LIBRARY_PATH=/usr/local/lib64
```

To the main .sh script. I'm interested in your addon because it has an auto feature. 

Here is a copy of my system information. It is running on a Raspberry Pi 4 Model B / 8GB
## System Health

version | 2020.12.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.4.79-v8
arch | aarch64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4996
Installed Version | 1.9.0
Stage | running
Available Repositories | 767
Installed Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | stable
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 28.6 GB
disk_used | 9.2 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.2.0), Samba share (9.3.0), ESPHome (1.15.3), Terminal & SSH (8.10.0), Home Assistant Google Drive Backup (0.102.0), RTL_433 to MQTT Bridge (0.2), Mosquitto broker (5.1), rtl_433 (0.1.0), rtl_433 MQTT Auto Discovery (0.1.4)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | storage
views | 7
resources | 2

</details>

Thank you for your help!

Nick

pbkhrv, 2020-12-31:
Thanks for the very helpful feedback! I added the environment variable through the Docker file instead of run.sh.

I just committed an updated version which should hopefully fix the issue.

Please click "Supervisor" -> "Add-on Store" -> "..." -> "Reload", then you should see an update for the rtl_433 add-on.

Let me know whether it fixes your issue.

Thanks!

nickF007, 2020-12-31:
Hi!

This is my log output now.

```
That was fast! [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf
rtl_433 version 20.11-12-ga20cd1a6 branch master at 202012232126 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Publishing MQTT data to homeassistant.local port 1883
Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
Registered 1 out of 176 device decoding protocols [ 15 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
```

I dont Know what 
```
[R82XX] PLL not locked!
```
means yet, but i'am going to play with the tool and find out if everything works!

Thanks!

pbkhrv, 2020-12-31:
"PLL not locked" is a normal message, even though it does look like a warning: https://www.rtl-sdr.com/forum/viewtopic.php?p=1101#p5615

pbkhrv, 2020-12-31:
You most likely already know about this, but just in case: I found "MQTT Explorer" to be a great way to check that rtl_433 publishing data as expected. Especially helpful when trying to understand how MQTT auto discovery works.

nickF007, 2020-12-31:
Nice! i downloaded MQTT explorer. It gives me information if i send it from the broker but no information from the addon. I must say i am totaly new to MQTT brokers..... i am not even a good programmer so.. it al pretty new and dificult for me!

But... ass i was typing this message this happend and i am happy!
[image]

Now i need to figere out how to put it in an entity and make everything work.... next puzzel for me!

Thank you a lot for your help!

pbkhrv, 2021-01-01:
Happy New Year! Sounds like the library issue is solved - I'll close this issue if you don't mind.

As for MQTT auto discovery, I just checked the rtl_433 script, and it doesnt look like it recognizes any of the keys that show up in your screenshot (command, dim, dim_value etc), unfortunately. There's still a way to do this using MQTT device triggers if you are trying to use the switch to send commands to HA, or to parse something like "dim_value" into a sensor. HA community forum might be the best place to get help with that stuff.

closed 2021-01-01 by pbkhrv