pbkhrv/rtl_433-hass-addons #3: Listen to multiple specific protocols at once

michthom, 2021-01-25:
I took a shot at refreshing the rtl_433_mqtt bridge a while back but never shared it publicly. In my attempt I wanted to give the option to specify a list of protocols to listen for, as in my case I want to pick up readings from my oil tank and also from the electricity monitor as well.

I may be missing something but is the current rtl_433 -c (config file) approach limited to a single protocol selection?
In my case I resorted to parsing the config so I could feed in independent -R options to select the protocols of interest.

If my code is of any use please feel free to grab it? I've sent you an invite to collaborate so you can see the code as I left it here: https://github.com/michthom/hassio_addons

pbkhrv, 2021-01-27:
Hi Michael, thanks for reaching out. I just ran a test, and rtl_433 does support having multiple frequencies specified via the config file - the only caveat is that it can listen to one frequency at a time for some time, then it switches, and goes around and around the list of frequencies (guessing because the hardware radio tuner can only tune to one frequency at a time).

Here's a snippet from my config file:
```
frequency   315M
frequency   433.92M
convert     si
report_meta newmodel
hop_interval  10
```

And here's the log output:
```
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 315.000MHz.
MQTT Connected...
MQTT Connection established.
Tuned to 433.920MHz.
Tuned to 315.000MHz.
```

The "hop_interval" is important - if you don't specify it, by default it'll listen to the first frequency for 600 seconds before switching to the next one. I set it to 10 seconds, and can see those "Tuned to ..." log entries piling up.

michthom, 2021-01-27:
Hi Peter - we have crossed wires - I was asking about limiting the protocols that the rtl_433 tool listens for rather than listening on multiple frequencies. I've got a number of devices all *on the same frequency* (433.92, boring!)  but I only want to listen to messages from some of them. Possibly a unique use case/personal quirk, so I'd understand if this isn't something you want to work on.

In my attempt, I allowed a list of desired protocols to be specified in the add-on config, which I then expand into individual -R arguments in the call to rtl_433, as I couldn't see a way to specify this in a monolithic config file...

```RTL_PROT=""
if ! bashio::config.is_empty "rtl433_protos"; then
  for protocol in $(bashio::config 'rtl433_protos'); do
    RTL_PROT=${RTL_PROT}"-R "${protocol}" "
  done
fi

/usr/local/bin/rtl_433 \
    -F mqtt://${MQTT_HOST}:${MQTT_PORT},user=${MQTT_USER},pass=${MQTT_PASS},retain=1,devices=${MQTT_ROOT}[/model][/id] \
    -M newmodel \
    ${RTL_PROT} \
    || bashio::log.error "rtl_433 process has stopped!"
```

pbkhrv, 2021-01-29:
Ah, I see. I'm actually limiting the protocols too - it works very simply via the config file, just list out protocol numbers one per line, like this:
```
protocol    40
protocol    86
```

This is what I see in the log as a result:
```
...
Registered 2 out of 176 device decoding protocols [ 40 86 ]
...
```

michthom, 2021-01-29:
D'oh! Sorry - I hadn't even thought to try that, I'll close this one and thanks for the explanation!

closed 2021-01-29 by michthom