pbkhrv/rtl_433-hass-addons #4: Add-on not working with error:  usb_open error -1

telegreen, 2021-02-09:
Since last updates of HASS.IO OS and supervisor on my both instances (Rpi3 x64 and VMware) add-on is not working with the same errors: 

Log: 

Publishing MQTT data to 192.168.72.109 port 1883
Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
Registered 148 out of 178 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-178 ]
usb_open error -1
[cmd] /run.sh exited 2

Core
Version
core-2021.2.2

Supervisor
Version
supervisor-2021.02.5

Operating System
Home Assistant OS 5.11

Everything was excellent before last updates!

schorsch3000, 2021-02-10:
It's the same for me, if there is anything i can provide to help, i'm here.

Also in my supervisor-log there is:
`21-02-10 19:07:02 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'devices' use a deprecated format, the new format uses a list of paths only. Please report this to the maintainer of rtl_433
`

pbkhrv, 2021-02-10:
Thanks a lot for reaching out, both of you. Investigating.

pbkhrv, 2021-02-10:
I changed the add-on config to match the [docs](https://developers.home-assistant.io/docs/add-ons/configuration/#add-on-config). It's working on my instance. Please test and let me know whether it works for you. Thanks!

schorsch3000, 2021-02-10:
everything is fine again, thank you very much for that blazing fast fix!

telegreen, 2021-02-11:
I confirm everything is ok on my both instances too!  Thank you for fix! 

Add-on version - [0.1.2] - 2021-02-10
Update add-on config to require USB access per new supervisor config format (thanks to @telegreen and @schorsch3000)

Issue closed.

closed 2021-02-11 by telegreen