pbkhrv/rtl_433-hass-addons #8: (0.1.2) MQTT connected -> connection failed -> connected ..

bommel947, 2021-05-19:
Hi, 

When I launch the addon (0.12) I see that it is successfully connecting to the MQTT broker.
But right after that it says "MQTT error 5" - which I suppose is an authentication problem.
It stays in the loop "connected -> connection failed -> connected ..."

Needless to say that the credentials are verified (mqtt lens)

Any help is very much appreciated!
Thank you!

The log shows me:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf
rtl_433 version 21.05-1-ga773fd0a branch master at 202105091257 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Publishing MQTT data to 192.168.0.21 port 1883
Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
MQTT Connected...
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
MQTT Connection error: 5
MQTT Connection failed...
MQTT Connected...
MQTT Connection error: 5
MQTT Connection failed...
MQTT Connected...
MQTT Connection error: 5
MQTT Connection failed...
MQTT Connected...

pbkhrv, 2021-05-25:
To eliminate the possibility that incorrect credentials are being supplied, can you please use deliberately incorrect MQTT information in your config file and see whether you get the same behavior?

Ninja1283, 2021-06-11:
I had a similar issue when setting up the add-on recently. Regardless of the credentials, it would always give me the connection error regardless of whether or not credentials were correct. I had a single quote (') in my mqtt password, and tried enclosing the password in double quotes, but it still would give me the connection error. Once I changed to an alpha-numeric password, everything worked fine.

pbkhrv, 2021-06-16:
> Hi,
>
> When I launch the addon (0.12) I see that it is successfully connecting to the MQTT broker.
> But right after that it says "MQTT error 5" - which I suppose is an authentication problem.
> It stays [...]

Looking at ninja1283's comment above - does your MQTT password contain a single quote character by any chance?

bommel947, 2021-06-16:
Nope. The pwd does not contain a single quote character

gdlesage, 2021-07-16:
I had same issue as bommel947, but I found when I removed the pound sign '#' from my password, MQTT connection was successful.  Looking at the log file, the message would truncate at the # suggesting the pound sign was being interpreted as the beginning of a python comment

bommel947, 2021-07-16:
Yeah. That could bei the reason. I also have a pound sign in my password!!

Ninja1283, 2021-07-28:
@deviantintegral and @pbkhrv, while you're troubleshooting the proper escapes for special characters, would it also be possible to add the ability to use data stored in the Home Assistant secrets file?

jbagdonas, 2021-10-05:
I had a similar problem.  Using a dash in the mqtt broker's username made the service fail to connect to the broker with the same errors:
MQTT Connected...
MQTT Connection error: 5
MQTT Connection failed...

Example, using a username of "mqtt-service" causes this problem to occur.  A username without the dash "mqttservice" does work though.

SxthGear, 2022-01-14:
Having the same issue here. No special characters (alphanumeric only) in the username or password. Looking at the MQTT/Mosquitto broker log it's successfully connecting and significantly spamming the host.

> 1642195512: New connection from 172.30.32.1 on port 1883.
1642195513: New connection from 172.30.32.1 on port 1883.
1642195513: New connection from 172.30.32.1 on port 1883.
1642195513: New connection from 172.30.32.1 on port 1883.
1642195514: New connection from 172.30.32.1 on port 1883.
1642195515: New connection from 172.30.32.1 on port 1883.
1642195516: New connection from 172.30.32.1 on port 1883.
1642195517: New connection from 172.30.32.1 on port 1883.
1642195517: New connection from 172.30.32.1 on port 1883.
1642195517: New connection from 172.30.32.1 on port 1883.
1642195517: New connection from 172.30.32.1 on port 1883.
1642195517: New connection from 172.30.32.1 on port 1883.
1642195517: New connection from 172.30.32.1 on port 1883.
1642195518: New connection from 172.30.32.1 on port 1883.
1642195519: New connection from 172.30.32.1 on port 1883.
1642195520: New connection from 172.30.32.1 on port 1883.
1642195521: New connection from 172.30.32.1 on port 1883.
1642195522: New connection from 172.30.32.1 on port 1883.
1642195523: New connection from 172.30.32.1 on port 1883.
1642195524: New connection from 172.30.32.1 on port 1883.
1642195525: New connection from 172.30.32.1 on port 1883.

Also, the feature request in #18 does not seem to work as it tries to parse the secrets "${username}" format as the actual configuration values.

deviantintegral, 2022-03-27:
I believe this is fixed since the `0.1.3` update to support multiple radios. If there are still issues with `0.2.1` or newer, please feel free to reopen. Thanks!

closed 2022-03-27 by deviantintegral