pbkhrv/rtl_433-hass-addons #9: Auto discovery, add retain flag ?

wreality, 2021-05-24:
Can the autodiscovery addon be updated to provide a config value to set the retain flag ?  The python script has a parameter available, but its currently not configurable from the addon config.

```
parser.add_argument("-r", "--retain", action="store_true")
```

Just plugged in and spun up both addons tonight and they worked fantastic right out of the box.  Thanks for the work to get this packaged into an addon... saved me all sorts of headaches!

pbkhrv, 2021-05-25:
looking at it now

pbkhrv, 2021-05-25:
new version 0.1.8 of the add-on is out - it allows you to pass "--retain" flag to the autodiscovery script. Please update.

wreality, 2021-05-25:
Works perfectly.  Thanks for the quick update!

closed 2021-05-25 by wreality