pbkhrv/rtl_433-hass-addons #10: if "model" not in data

docsparks, 2021-05-25:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting rtl_433_mqtt_hass.py...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3452, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1779, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1181, in loop
    rc = self.loop_read(max_packets)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 2310, in _packet_read
    rc = self._packet_handle()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 2936, in _packet_handle
    return self._handle_publish()
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3216, in _handle_publish
    self._handle_on_message(message)
  File "/usr/lib/python3.8/site-packages/paho/mqtt/client.py", line 3444, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/rtl_433_mqtt_hass.py", line 481, in mqtt_message
    bridge_event_to_hass(client, topicprefix, data)
  File "/rtl_433_mqtt_hass.py", line 541, in bridge_event_to_hass
    **if "model" not in data:**
TypeError: argument of type 'int' is not iterable

please let me know if you need to know anything else.
thanks

pbkhrv, 2021-06-07:
still investigating. can't reproduce the error, and by the looks of the Paho MQTT library, this is not supposed to happen...

pbkhrv, 2021-06-07:
@docsparks Is that error intermittent? does the addon work at all? Wondering whether that message with payload that looked like a number instead of a proper JSON object was a fluke or it happens every time. Where did you copy that JSON message from, BTW?

docsparks, 2021-06-11:
mqtt json is from the log of the rtl433 plugin.  the error message i post is from the autodiscovery plugin.

I was able to create a sensor via standard mqtt methods however.

# rtl_433/9b13b3f4-rtl433/devices/Fineoffset-WH51/00c7ae/moisture
  - platform: mqtt
    name: "Soil Meter"
    state_topic: "rtl_433/9b13b3f4-rtl433/devices/Fineoffset-WH51/00c7ae/moisture"
    force_update: true

pbkhrv, 2021-06-16:
Thanks for more info. I'm sorry that the autodiscovery add-on didn't work for you. Let's leave this bug report open for now and see if anyone else has this problem, or can contribute more data so that I can understand the problem.

deviantintegral, 2022-01-12:
Would you consider filing a pull request upstream for this? https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py is where this lives.

deviantintegral, 2022-06-12:
Hm, I thought the default topic is `rtl_433/+/events`. I don't think we support using anything other than an "event" item, which is what using `rtl_433/#` would subscribe to.

The docs of course say `rtl_topic: MQTT topic where rtl_433 is publishing its output. Default is "rtl_433".`. This may be something we need to clean up in the docs on our end.

deviantintegral, 2024-02-17:
Given this hasn't come up in over a year, and also would be an upstream fix if anything, I'm closing this out. If there's something to do in this addon please feel free to file a new issue.

closed 2024-02-17 by deviantintegral