pbkhrv/rtl_433-hass-addons #16: reset rain gauge to zero

gdlesage, 2021-07-16:
First comment is great work, and easy to set up
I noticed, which maybe be by design, that I am unable to reset  Acurite rain gauge (Acurite-Rain899-0-1322-RT) rain value back to zero.   I used mqtt explorer to reset topic 'rtl_433/9b13b3f4-rtl433/devices/Acurite-Rain899/0/1322/rain_mm' to zero, but within a min or so the topic valve returns to previous value for accumulated rainfall.  This is true with rtl_433 addons running or not running, if not running, when I restart them, the rainfall changes from 0 to previous accumulated value.  Over time I can imagine this number gets quite large.  I can use utility meter home assistant integration to monitor interval rainfalls.  Is there a trick to reset to zero?

deviantintegral, 2021-07-16:
Interesting - what is the rain gauge itself sending? Can you check by adding `output kv` to your config? In general, some types of sensors have an incrementing counter that is only reset when they lose power. For example, in influxdb there's the `non_negative_derivative` function for this: https://stackoverflow.com/questions/38016051/how-do-i-get-consistent-values-with-influxdb-non-negative-derivative.

gdlesage, 2021-07-16:
forgot to mention that when  I reset rain gauge physical base station to zero rain mm  HA still shows total accumulation rather than reset to zero.   

Adding out kv gives me the following to log output:

time      : 2021-07-16 13:20:02
model     : Acurite-Rain899                        id        : 1322
channel   : 0            Battery   : 0             Rainfall Accumulation: 82.55 mm                     Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-16 13:21:02
model     : Acurite-Rain899                        id        : 1322
channel   : 0            Battery   : 0             Rainfall Accumulation: 84.58 mm                     Integrity : CHECKSUM

between 13:20:02  and 13:21:02 time, I tilled to rain gauge from side to side to increment the rain measurement.

gdlesage, 2021-07-16:
but you are right deviantintegral , removing the batteries resets the reading to zero

time      : 2021-07-16 13:31:03
model     : Acurite-Rain899                        id        : 1322
channel   : 0            Battery   : 0             Rainfall Accumulation: 0.00 mm                      Integrity : CHECKSUM

case closed

closed 2021-07-16 by gdlesage