pbkhrv/rtl_433-hass-addons #18: Feature Request: Compatibility with secrets.yaml

Ninja1283, 2021-07-28:
Somewhat related to #8, are you able to add compatibility with the Home Assistant secrets file to avoid having plaintext credentials stored in the config?

deviantintegral, 2021-07-28:
To do that, I think we'd need to parse the `secrets.yaml` file (with python?), but then we'd end up having to write them back out to disk (following the work at https://github.com/pbkhrv/rtl_433-hass-addons/pull/14). Or, we'd have to use the `-F` option instead when starting the process (as it does look like rtl_433 excludes the username and password settings from the process listing).

deviantintegral, 2022-01-12:
We no longer require secrets to be stored in the config, as long as you're using the MQTT broker within Home Assistant. Does that solve this need for you?

Ninja1283, 2022-01-15:
> We no longer require secrets to be stored in the config, as long as you're using the MQTT broker within Home Assistant. Does that solve this need for you?

The documentation still says that "the absolute minimum that you need to specify in the config file is the MQTT connection and authentication information: `output      mqtt://HOST:PORT,user=XXXX,pass=YYYYYYY`"

Is that not still the case?

deviantintegral, 2022-01-17:
No, it's not, and we should improve that text.

> and an external MQTT server location

Yes, that's what I was trying to get at - there's no way for us to autodetect those credentials. secrets.yaml would be a nice improvement for those setups, and while I don't expect to prioritize it myself (since I wouldn't use it!) I'm glad to review and test PRs.