pbkhrv/rtl_433-hass-addons #19: Feature Request: rounding of values

booroondookZ, 2021-08-10:
Now the add-on sends values with many decimal digits after the dot.
For example, temperature is 28,83334 °C.
But in practice, I don't need that precision.
It would be nice if there was a rounding option in the settings indicating how many decimal places should be.

realthk, 2021-11-22:
Yes, the same with wind-speed:
[image]

It's easy to create a rounded template sensor in HA, but it would be much better to handle in RTL 433. It even could be a global option, as I guess a 2 decimal rounding won't hurt any sensor.

deviantintegral, 2021-11-25:
https://github.com/merbanan/rtl_433/issues/1509 discusses this upstream. It's pointed out that different systems consuming the MQTT data may need different precision to parse the float correctly. However, in our case, we know that we're consuming the data in Home Assistant and could make some assumptions.

If we implement rounding, would we do it for all floats, or just specific floats? What if different devices needed different options for rounding?

Perhaps more interesting, what if Home Assistant or some upstream library is parsing the float incorrectly? I'm assuming you've confirmed that the numbers in Home Assistant match the MQTT data?

I'm happy to continue the discussion here for now, but I think this is likely best as an update to the upstream script at https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py.

realthk, 2021-11-25:
Oh, I see, and already found the bug there, [this conversion](https://github.com/merbanan/rtl_433/blob/c85279804c8231a49da64a0820c37657b39f1b27/examples/rtl_433_mqtt_hass.py#L269) is wrong:
`"value_template": "{{ float(value|float) * 3.6 | round(2) }}"`

Because in Jinja templating of HA, the order of execution is not guaranteed to be left-to-right, only if you use parentheses around everything before the last filter like this:
`"value_template": "{{ (float(value|float) * 3.6) | round(2) }}"`

For example, copy this into the template editor of developer tools in HA, and get the same 1.4400000000000002:
`{{ float(0.4|float) * 3.6 | round(2) }}`
while this results the correct 1.44:
`{{ (float(0.4|float) * 3.6) | round(2) }}`

Now of course this only affects the lines having round() filter in the linked rtl_433_mqtt_hass.py, like speed conversions from m/s to km/h, calculation rain rate/hour, not the temperature, which is simple `{{ value|float }}` - but I have no problem with my RTL 433 temperature values in HA, that might be something else with the problem of booroondookZ.

closed 2021-11-25 by booroondookZ

deviantintegral, 2021-11-25:
Please link the issue here when you do.

realthk, 2021-11-25:
[I did](github.com/merbanan/rtl_433/issues/1888), and as the fix is very simple, it was merged at once