pbkhrv/rtl_433-hass-addons #23: Bug: rtl_433 Alpine:latest code doesn't work on RPi4

BarryMar, 2021-10-25:
Hi there.

I've been trying to see if I can get a  Nooelec NESDR SMArTee v2  RTL2832U ( https://www.nooelec.com/store/sdr/sdr-receivers/smart/nesdr-smartee.html)  to work with rtl_433-hass-addon on my Raspberry PI4 to interpret my LaCrosse Technologies LTV-TH2 Weather Sensor 915 MHz output.  

I was able to get good data with rtl_433 on my Mac with code from Homebrew, so I knew it was theoretically possible to do.  Unfortunately, every time I tried to use rtl_433-hass-addon it ended immediately in "[cmd] /run.sh exited 1".

Trying the rtl_433 Docker code used by this addon on a straight Raspberry PI 400 with latest Raspberry PI OS 32-bit, the Alpine code would instantly die with an "Assertion failed" error.  

According to https://github.com/hertzg/rtl_433_docker/issues/3 this is a known problem with ARM Alpine 3.13 in Latest caused by some stale library code in various distributions.   There are two possible solutions:

1.     Use the alpine-3.12 version based images (example: hertzg/rtl_433:alpine-3.12-latest)
2.     Use debian based images (example: hertzg/rtl_433:debian-buster-latest)

After I figured out how to get into my real HAOS Root with https://github.com/adamoutler/HassOSConfigurator , I was able to get the code to run properly in this environment with the same command as the P400:

docker run --device /dev/bus/usb/001/003 hertzg/rtl_433:debian -f 915M

I'm still learning the basics of Docker at the moment, so I don't know what's all possible to do with it or how to reconfigure the files from this codebase yet.  

Is it possible to make the rtl_433-hass-addon Docker initialization code branch for ARM processors to use the Debian build instead of Latest?

deviantintegral, 2021-11-10:
I have a Raspberry Pi 4 as well, but it has the 64 bit version of HAOS on it. Testing any PRs on 32-bit would be a stretch for me right now, but I could verify it doesn't break on x64 and arm64.

I won't likely be driving any changes around this, but I'm glad to review PRs.

deviantintegral, 2022-01-12:
Since I wrote the above, I had the opportunity to learn about cross-architecture Docker builds. I was working with arm64, but I think there should be a way to test this without needing a second Raspberry Pi.

deviantintegral, 2022-03-27:
> I have a Raspberry Pi 4 as well, but it has the 64 bit version of HAOS on it

I totally got this wrong, I actually have 32-bit HAOS installed there. The latest containers have been working fine on that system. My guess is this has been fixed upstream, or fixed by ensuring that all systems pull the same container image. Feel free to reopen if this is still an issue in `0.2.1` or higher. Thanks!

closed 2022-03-27 by deviantintegral