pbkhrv/rtl_433-hass-addons #25: light_lux

gabrieledaltoe, 2021-11-02:
Published Fineoffset-WH65B/59: battery_ok, temperature_C, humidity, wind_dir_deg, wind_avg_m_s, wind_max_m_s, rain_mm, uv
Skipped Fineoffset-WH65B/59: uvi, light_lux

Here MQTT-EXPLORER:
{"time":"2021-11-02 11:52:26","model":"Fineoffset-WH65B","id":59,"battery_ok":1,"temperature_C":14.7,"humidity":72,"wind_dir_deg":196,"wind_avg_m_s":2.10375,"wind_max_m_s":2.55,"rain_mm":6316.72607,"uv":1188,"uvi":2,"light_lux":45012.0,"mic":"CRC"}

can you update script to look "light_lux" ?

GeorgeIoak, 2022-01-02:
+1, looks like the script is expecting lux to be formatted as lux when RTL_433 sends light_lux. Here's from my unit:

```
{"time":"2022-01-01 17:52:41",
"model":"Cotech-367959",
"id":209,
"battery_ok":1,
"temperature_C":5.38889,
"humidity":69,
"rain_mm":333.89999,
"wind_dir_deg":359,
"wind_avg_m_s":0,
"wind_max_m_s":0,
"light_lux":0,
"uv":0,
"mic":"CRC"}
```

It's currently at night which is why the value is 0

GeorgeIoak, 2022-01-02:
I went into the container and modified `rtl_433_mqtt_hass.py` from 
```
    "lux": {
        "device_type": "sensor",
        "object_suffix": "lux",
        "config": {
            "name": "Outside Luminancee",
            "unit_of_measurement": "lux",
            "value_template": "{{ value|int }}"
        }
    },
```

to:
```
    "light_lux": {
```

I confirmed that this worked to bring in the lux values from the weather station. I think my manual changes in the container will get wiped out with a HA restart so this should be changed.

I can do a PR if that makes it easier to update the code @pbkhrv

deviantintegral, 2022-01-02:
@GeorgeIoak can you file a PR up at https://github.com/merbanan/rtl_433/ ? That's where this file originally comes from.

GeorgeIoak, 2022-01-02:
[PR](https://github.com/merbanan/rtl_433/pull/1930) has been submitted

GeorgeIoak, 2022-01-03:
and successfully merged, https://github.com/merbanan/rtl_433/pull/1930

GeorgeIoak, 2022-01-12:
Just curious how this process works. I see the changes made back at rtl_433 but they haven't issued a new release. Looks like you checkout specific releases and don't use the latest repo so does this mean this change won't hit until rtl_433 issues a new release and then you'll have to issue a new release as well?

deviantintegral, 2022-01-12:
There's been enough interest in newly merged features and bug fixes that I'm thinking of changing from tracking tags, to tracking specific commits on the upstream master branch.

Or... I just realized, nothing stops us from building _two_ or more copies of `rtl_433` and then adding a configuration option to choose the version. The default would be the last tag, and the latest would be whatever version is on master at the time. I'll file an issue for that!

glendower, 2022-01-16:
I still have this problem and I installed the two RTL_433 files a couple of days ago. 

Published Cotech-367959/72: time, battery_ok, temperature_F, humidity, rain_mm, wind_dir_deg, wind_avg_m_s, wind_max_m_s, uv
Skipped Cotech-367959/72: light_lux

I also have problems with CurrentCost devices

Published CurrentCost-TX/849: time
Skipped CurrentCost-TX/849: power0_W, power1_W, power2_W

and my other weather station Fineoffset

Published Fineoffset-WHx080/0/122: time, battery_ok, temperature_C, humidity, wind_dir_deg, wind_avg_km_h, rain_mm
Skipped Fineoffset-WHx080/0/122: wind_max_km_h

How can I fix these problems ?

GeorgeIoak, 2022-01-17:
The fix is in place back in the rtl_433 repo but this add on pulls a specific release. You can manually make a change now to pull the latest rtl_433 repo or wait until @deviantintegral makes some changes so his code allows tracking from commits.

glendower, 2022-01-17:
Thankyou, how do I do that ?

GeorgeIoak, 2022-01-17:
Probably the easiest would be to fork this repo and then edit the [lux definition](https://github.com/pbkhrv/rtl_433-hass-addons/blob/927fd4101e09bb997010db7da344496472549dd1/rtl_433_mqtt_autodiscovery/rtl_433_mqtt_hass.py#L438)

Then you can add your own repo to HACS

The other option is to go into the container running this and edit the same file. The downside to do doing that is the change isn't permanent since a restart of HA will recreate the container pulling the source from here

glendower, 2022-01-17:
I did try to do this before I posted but had no idea how to do it.

Having looked up the meaning of 'fork' I have managed to edit the rtl_433_mqtt_hass.py file but no idea what to do next.

Can you give a few pointers on how I proceed after the edits ?

GeorgeIoak, 2022-01-17:
So I assume that you forked this repo and made the changes in your forked repo. Now you just go back and [install just like you did for this repo](https://www.home-assistant.io/common-tasks/os#installing-third-party-add-ons) but instead of using the URL for this repo you use the URL for your repo. You can leave this repo installed but just stop it in HA by going to the Add-on page, click on the repo tile and then click on stop.

glendower, 2022-01-18:
Thankyou so much for your help and patience

Lux is coming through loud and strong, This has taken my knowledge base to a whole new level and I feel able to tackle the other two issues, Currentcost power and wind max on the fineoffset weather station.

The problem  of UTC time I'll leave for another day.

Thanks again

GeorgeIoak, 2022-01-18:
Glad you got it straightened out, good job!

deviantintegral, 2022-02-12:
We're currently pulling in a near-latest commit from master for `rtl_433`. The next release will fix this specific issue, and it's been merged into the `next` branch.

closed 2022-02-12 by deviantintegral