pbkhrv/rtl_433-hass-addons #27: HA MQTT: rtl_433_mqtt_hass.py: error: argument -H/--host: expected one argument

ecylcje, 2021-11-09:
Hi,

I'm not exactly sure when this happened but my mqtt auto discovery addon has stopped working. It is now failing below:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[13:28:10] ERROR: Got unexpected response from the API: Service not enabled
Starting rtl_433_mqtt_hass.py...
usage: rtl_433_mqtt_hass.py [-h] [-d] [-q] [-u USER] [-P PASSWORD] [-H HOST]
                            [-p PORT] [-r] [-f] [-R RTL_TOPIC]
                            [-D DISCOVERY_PREFIX] [-i DISCOVERY_INTERVAL]
rtl_433_mqtt_hass.py: error: argument -H/--host: expected one argument
[cmd] /run.sh exited 2
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

I have noticed that the configuration has changed and no longer seems to require the username/password/server fields:

mqtt_host: XXXXX
mqtt_port: 1883
mqtt_user: XXXX
mqtt_password: XXXX

Apologies if this is really simple!

Chris

deviantintegral, 2021-11-09:
You may need to reload the integration from the addons list overflow menu. Otherwise, I think Home Assistant won't read the updated `config.json`.

ecylcje, 2021-11-09:
Thank you so much for your quick reply.

I'm not sure what you actually mean by the overflow menu? I've actually completely uninstalled the 'rtl_433 MQTT Auto Discovery' addon and then reinstalled it from scratch. I've also restart HomeAssistant.

I'm guessing the issue is that:

MQTT_HOST=$(bashio::services mqtt "host")

isn't returning any data. Is there a place I should be enabling the supervisor API to allow this addon to get the details from my MQTT install? Should I uninstall that integration and re-install?

Thanks again,

Chris

einarjh, 2021-11-09:
This is broken for me too. I tried going to the addon store, clicking on the three dots in the upper right corner and "reload", but the service is still broken.

einarjh, 2021-11-09:
Additionally, I had to do a full uninstall before restoring the backup to get it working again. I don't know how the new version is meant to work, but it didn't seem to be picking up the correct mqtt host, port and other details correctly for me.

deviantintegral, 2021-11-10:
Unfortunately I don't see a way to add an addon repository, but then install an older version to try to replicate this.

What I did do was:

1. Uninstall both addons.
2. In the "addons" share exposed by the Samba addon (which can also be accessed via the Terminal addon), I checked out 982f44d from this repo, which represents the code as of July 2021.
3. Installed both addons and configured them.
4. Checked out `main` which is currently f98beb9
5. Pressed the reload option in the addons UI:
 
[image]

6. This dialog showed up in the addon config:
 
[image]

7. I clicked "update", the addon restarted and the config screen updated (removing the manually set user and password).

>  I don't know how the new version is meant to work

The idea is to automatically use the HASS API to autodetect the mqtt hostname and credentials from the Mosquitto addon. But, looking at the code that was actually written:

`"services": ["mqtt:want"]`

I think that may need to be `need` instead. Does anyone have a good case for why the MQTT server would not be using the Mosquitto addon? It strikes me a little odd to have the autodiscovery script inside HA, just to add topics to an external MQTT server, just to re-consume that back into HA, but perhaps there's good reasons to do that.

ecylcje, 2021-11-10:
Thanks for all the info.

Are we saying that at the moment this addon is broken, but will be fixed eventually - and I should just wait? Or is the idea to try and revert back to 0.1.8 version and stick with that?

Thanks again,

Chris

deviantintegral, 2021-11-10:
@ecylcje are you using the Mosquitto addon in Home Assistant, or an external broker? If it's the addon, it should be working, and I'd like to dive into why it isn't.

ecylcje, 2021-11-11:
Hello, I was using this addon:

The MQTT integration was introduced in Home Assistant pre 0.7, and it's used by 37.5% of the active installations.

I did a search for Mosquitto and it wasn't listed in the available integration. I checked on the MQTT and it is working fine and I can search the MQTT queue.

Have I missed installing something?

Thanks,

Chris

deviantintegral, 2021-11-12:
Ah, I'm referring to the [official mosquitto addon](https://github.com/home-assistant/addons/tree/master/mosquitto), which then is exposed to Home Assistant through the MQTT integration. If you're not familiar with it, you must have an MQTT broker installed otherwise.

I will aim to at least get a PR up supporting manual host / user / password / ports again up in the next day or so. In the short term, feel free to downgrade to the previous release. Sorry for the hassle.

ecylcje, 2021-11-12:
Understood. Yes I just installed the 'mosquitto broker' you mentioned and the RTL 433 addon now starts.

But you are absolutely right. I have a mosquitto broker installed on another Raspberry PI etc so I don't want to use the HA mosquitto broker. Hopefully you can make the system support both the HA broker and an external broker.

Thanks so much,

Chris

deviantintegral, 2021-11-13:
I've filed a PR for this, but have not fully tested with an external broker. Please give it a try and post any issues.

ecylcje, 2021-11-16:
I do apologies for not replying, but I'm not sure of the way to test this PR. Is there a method for checking it out etc and doing the install automatically?

deviantintegral, 2021-11-17:
@ecylcje Do you have the Samba or Terminal addon setup? If so, you can download https://github.com/pbkhrv/rtl_433-hass-addons/archive/refs/heads/mqtt-manual.zip and extract it into the addons folder, and then reload addons. It will show up in a "Local" section. I believe that will override addons added via "Repositories...", but if not try removing it from there too.

Worst case, I can take some time to set up an external broker and if that works you can test after merging.

ecylcje, 2021-11-17:
Hello,

 

Hope you don’t mind I just quickly forked your code with your PR include.

 

I then just deployed the ‘forked’ repo which contained your PR code and then tested that. Everything worked fine and the rtl433 items are now being correctly reported on the Overview page.

 

Thanks so much,

Chris

 

 

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.

[s6-init] ensuring user provided files have correct perms...exited 0.

[fix-attrs.d] applying ownership & permissions fixes...

[fix-attrs.d] done.

[cont-init.d] executing container initialization scripts...

[cont-init.d] done.

[services.d] starting services

[services.d] done.

Using an external mqtt broker.

Starting rtl_433_mqtt_hass.py...

MQTT connected: Connection Accepted.

deviantintegral, 2021-11-18:
I merged the PR and bumped the version. Thanks for testing!

PR #28 (2021-11-18): Restore options for external brokers
Resolves #27 .
closed 2021-11-18 by deviantintegral