pbkhrv/rtl_433-hass-addons #31: protocol causing crash

ideal2545, 2021-12-02:
Hi, 

I'm having a weird issue. So I recently did a 'rebuild' so that I would pull down the latest version of rtl_433, looks like it had some changes with how it handles conf files but I specified the exact conf file in the supervisor config and it loaded up just fine. 

I attempted to enable protocol 192 (its a newer protocol somewhat recently merged to master for Govee leak sensors) and when I do that, the container just exits. I'm having a hard time trying to figure out why. I did confirm the version of RTL running is the latest so maybe I'm missing something?

deviantintegral, 2021-12-03:
> its a newer protocol somewhat recently merged to master

That's it right there. We don't build off of master, but off of the last release which is still 21.05: https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/Dockerfile#L19

To run master, you can either clone this repo into your addons directory and edit that line, or log in to the supervisor (not HA!) command line and manually build the container.

ideal2545, 2021-12-04:
That makes sense, thanks a lot ill give that a go!

closed 2021-12-04 by ideal2545