pbkhrv/rtl_433-hass-addons #32: Causes HA to hard crash

hw-monkey, 2021-12-07:
First, when I installed this, I was so tthankful. It is everything I was hoping for.

After several weeks of running this on the HAOS with a SSD, with a NESDR Mini 2+, I have found it has been making my home assistant unstable. The system will _hard crash_ making it non-responsive to ethernet traffic. I cannot ssh into the device, it does not respond to pings, and sniffing traffic on the router port shows traffic being sent, but not responses. It even loses its IP address. A true hard crash that can only be recovered with a restart.

After restart, it will last for hours or up to two days.

Given the way logs clear, I am unable to see what happens. But, I do log device temperature, CPU use, and hard drive utilization. These three look normal.

What can I do to debug/capture information that I could send to you that might help determine what is going on? Keep in mind, this is HA on a Raspberry Pi 4, with 4G RAM, that is running the HAOS, so I am looking for a native solution.

Thanks!!!

deviantintegral, 2021-12-07:
Ouch, that's a pain to debug. Is it possible to have a monitor plugged in so you can see any kernel crash messages?

I think it's unrelated, but I will mention it just in case. I have 2 HA installs, and one is also on the same model of Raspberry Pi. I was having trouble where the Pi would cause a network loop and bring down the entire LAN (and itself in the process). I switched it to wifi, and it's been fine ever since. I had it chalked up to a bad network cable, and didn't have one at the time to replace it. But - I had switched ihe Pi to ethernet at the same time as installing the NESDR Mini 2+. So, perhaps running on wifi for a bit could be a good test?

hw-monkey, 2021-12-08:
I connected it to a monitor last night. So far, no crash.

My network is pretty robust. I have spanning tree enabled, so a loop like that would be shut down, and the router would highlight it. But, that may be the next thing to watch

hw-monkey, 2021-12-09:
It did crash. And, amazingly, I was able to see it in time to catch some messages. I did not catch the beginning, but it started flooding the log file. Then, even though it had ssd space (and does not have an sd card), it freaked out. SO, I went to adjust the logging detail level, and I saw that I had inadvertently left HA on debug mode. I changed it back to critical. Now, let's see what happens.

Messages were basically....
```
systemd-journald[114]4Failed to write entry ... read only file system
var/log/journal/0c612bf4fe1xxx/system.journal
var/log/journal/0c612bf4fe1xxx/user-1003.journal
var/log/journal/0c612bf4fe1xxx/user-1004.journal
var/log/journal/0c612bf4fe1xxx/user-1001.journal
Failed to send WATCHDOG=1 notifcation message: Transport endpoint is not connected
```

deviantintegral, 2021-12-09:
> systemd-journald[114]4Failed to write entry ... read only file system

That feels like a hardware or cabling issue, because AFAIK nothing we do in the addon (or in HA itself) should cause that filesystem to become read-only.

Is your SSD on USB or something else? If the NESDR is causing something to break with USB, that could cause issues with USB attached storage.

hw-monkey, 2021-12-11:
I hear what you are saying. But I disabled debug mode, and it has been stable as a rock for days, which just did not happen when these extensions were activated and disable mode was on. I have no idea. I can guess. The amount of logging when debug mode is so high that when this module is activated, MQTT trips over itself at some point. And that trip causes the whole thing to freeze. Again, just a fellow coder who is guessing based on what I have seen.

Thanks again. Mark this as resolved!

closed 2021-12-11 by hw-monkey