pbkhrv/rtl_433-hass-addons #33: Can't see Govee Leak Detector messages

sweichbr, 2021-12-08:
Hi

I know that the Govee Water sensors have recently been added to rtl_433 (device 192), but for some reason I am not seeing the message come through mqtt.

My assumption is that this add on installs the latest version of rtl_433.  Is that the case, or is there some other cause for them not to show up?

I had previously run rtl_433 on a separate Raspberry PI, but now am attempting to get it running directly in HA.

deviantintegral, 2021-12-08:
See https://github.com/pbkhrv/rtl_433-hass-addons/issues/31

Looking at https://triq.org/rtl_433/CHANGELOG.html, it seems like releases are somewhat sporadic. If that's going to be the case in the future, it could make sense for us to switch to building from master. The downside is that it could mean that support of the addon is more difficult, as the same version of the addon could have widely varying rtl_433 releases inside.

I'm going to file an upstream issue to ask about release cadence.

deviantintegral, 2021-12-08:
Filed https://github.com/merbanan/rtl_433/issues/1902

ILAsoft, 2021-12-30:
Resurrecting -- it looks like the December 14th new release occurred that had this version in it (https://github.com/merbanan/rtl_433/releases/tag/21.12), yet I am not seeing it still - rtl_433 is reporting event leak_detected and Water Leak but Auto Discovery is only reporting (and skipping) test/time types eg below. Am I misconfiguring something on my end?

homeassistant/sensor/Govee-Water-61402/Govee-Water-61402-UTC/config : {"device_class": "timestamp", "name": "Govee-Water-61402-UTC", "icon": "mdi:clock-in", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/Govee-Water/61402/time", "unique_id": "Govee-Water-61402-UTC", "device": {"identifiers": "Govee-Water-61402", "name": "Govee-Water-61402", "model": "Govee-Water", "manufacturer": "rtl_433"}}
Published Govee-Water/61402: time
Skipped Govee-Water/61402: event, code

deviantintegral, 2021-12-30:
Long story short, we (I) need to get more consistent in terms of merging PRs and handling updates to https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/config.json#L3. As is, you won't get the new version of the rtl_433 container until that number is bumped or you uninstall and reinstall the addon (you can leave your configuration files untouched).

I've actually been running the 21.12 version for a while now and it seems fine. I'll bump the config file now.

deviantintegral, 2021-12-30:
Done!

It looks like you had already reinstalled, and were getting the latest release. The key I see is:

> Skipped Govee-Water/61402: event, code

That likely means that the autodiscovery script needs to be updated to handle those keys. In the short term, you can manually publish configuration to the homeassistant topic in mqtt to configure the entity.

If you or anyone else files a PR upstream, please leave a link here and I can at least give it a static review.

MikeAg03, 2022-02-20:
Any updates on this?

deviantintegral, 2022-03-27:
The `0.2.1` release was built from rtl_433's `master` branch as of early February. It should contain any device support up to then. If the leak detector still isn't showing up, feel free to reopen. Thanks!

closed 2022-03-27 by deviantintegral

esmoyer, 2022-08-23:
Really sorry to bring this up again, but searching Google I can't seem to figure this out. I am working on building a new HA instance and was giving this Add-in a try, it publishes the battery and a couple other entities but doesn't do the leak even, when I look at the log it looks like it's skipped, but for the life of me, I can't figure out why. I'm sure I am missing something stupid, any insight would be greatly appreciated!

[image]

deviantintegral, 2022-08-23:
To reduce mqtt traffic the autodiscovery script will not publish topics it has seen within a 10 minute window. Is it possible that the water meter publishes different messages at different times? In other words, instead of always transmitting all fields, it rotates or changes through them?

Can you set discovery_interval to 0 and see if that works?

esmoyer, 2022-08-23:
The Govee device seems to push out a "heartbeat" every so often, but then can force a button push on the device as well as manually trigger the device with water. But the add-on continues to skip any "event" message which would be the button press or the actual leak detected alarm. 
[image]

deviantintegral, 2022-08-26:
Yes, we can see from above that it's sending different fields at different times.

As I mentioned above, can you set discovery_interval to 0 and see if that works?

esmoyer, 2022-08-26:
> Yes, we can see from above that it's sending different fields at different times.
> 
> As I mentioned above, can you set discovery_interval to 0 and see if that works?

Sorry, didn't explicitly state; yes, the second post with the screenshot was a screenshot after setting the discovery_interval to 0 and restarting.

deviantintegral, 2022-09-02:
I forgot that "Skipped" is when it sees values that haven't been added yet. In this case, "event" is not defined in the autodiscovery script yet:

https://github.com/merbanan/rtl_433/blob/bbf9f34a2c73cccb225346ba03efb2604e8e9e3c/examples/rtl_433_mqtt_hass.py#L737-L752

Event needs to be added to the array at https://github.com/merbanan/rtl_433/blob/bbf9f34a2c73cccb225346ba03efb2604e8e9e3c/examples/rtl_433_mqtt_hass.py#L112-L116

If a PR is filed upstream, I'd be willing to patch it in here until it's merged.

deviantintegral, 2022-09-02:
Also, this is documented in the README for the addon at https://github.com/pbkhrv/rtl_433-hass-addons/blob/next/rtl_433_mqtt_autodiscovery/README.md#what-sensorsdevices-does-this-script-support

esmoyer, 2022-09-02:
That would explain it. I definitely misinterpreted that description of "event" (above the list of sensors) being synonymous with the Govee sensor "event" (that is skipped) but now realize the Govee sensor "event" belongs in the list with battery, temp, etc. (If I'm understanding it correctly)

I'm no coding expert but I may fork and try to toy around with it to maybe learn a bit more.

Thanks!