pbkhrv/rtl_433-hass-addons #34: All devices connected via rtl_433 have stopped working

P-etr, 2021-12-09:
Hi,

I would like to ask you for help with solving my problem.
Yesterday morning, all my devices connected via rtl_433 stopped working.
Home Assistant runs on Raspberry Pi 4B and I use this SDR https://www.passion-radio.com/sdr-receivers/rtl-sdr-r820t2-248.html

Here is the log from rtl_433:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms ... exited 0.
[fix-attrs.d] applying ownership & permissions fixes ...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts ...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting rtl_433 -c /config/rtl_433/rtl_433.conf
[13:51:32] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
rtl_433 version 21.05 branch at 202105091238 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Publishing MQTT data to core-mosquitto port 1883
Publishing device info to MQTT topic "rtl_433 / 9b13b3f4-rtl433 / devices [/ type] [/ model] [/ subtype] [/ channel] [/ id]".
Publishing events info to MQTT topic "rtl_433 / 9b13b3f4-rtl433 / events".
Publishing states info to MQTT topic "rtl_433 / 9b13b3f4-rtl433 / states".
Registered 2 out of 186 device decoding protocols [125 165]
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S / s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
MQTT Connected ...
MQTT Connection established.
```

Here is the log from rtl_433 MQTT Auto Discovery:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms ... exited 0.
[fix-attrs.d] applying ownership & permissions fixes ...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts ...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
mqtt found in this Home Assistance instance.
Starting rtl_433_mqtt_hass.py ...
MQTT connected: Connection Accepted.
```

I don't see any of the devices connected via rtl_433 in HA now.
I tried to connect SDR to a laptop with Windows and there I receive data from all devices via rtl_433 without any problem.
I think the problem will be in rtl_433 MQTT Auto Discovery, because I don't see in its log that any devices have been discovered. It used to be there.
As I tried to find out where the problem was yesterday, I restarted both add-ons several times. I also restarted the whole HA. Then the information from the device appeared to me a few times, but it was irregular and with outages, and now it doesn't work at all.
I also reinstalled both add-ons.
Yesterday, when I found out that rtl_433 is not working properly, its log also wrote `pulse_FSK_detect (): Maximum number of pulses reached`
It doesn't appear there now.

Here I attach the settings from rtl_433 MQTT Auto Discovery:

[image]

And here is the rtl_433 configuration file:

```
output      mqtt://core-mosquitto:1883,user=homeassistant,pass=XXXX
frequency   433.92M
protocol 125
protocol 165

```
Thanks to these add-ons, I can see in HA what the outdoor temperature is and also what volume of water remains in the well.
It worked fine the whole time and stopped abruptly yesterday.
I will be very happy for your advice.

deviantintegral, 2021-12-09:
Is it possible your credentials changed for MQTT? If rtl_433 can't write to MQTT, then there's no messages for the autodiscovery script to listen to either.

Note we now support automatic configuration of credentials since you're using the built-in mosquitto broker. I see you migrated your configuration files to the rtl_433 directory, but are hardcoding some values. Could you change your output line to:

`output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}`

And consider adding an additional `output kv` line, which will print all radio captures to the addon logs. But, I think it's likely your output line that's broken, and not something with the radio.

P-etr, 2021-12-09:
Thank you for the quick reply. I started playing with HA this year and I'm a beginner.

> Is it possible your credentials changed for MQTT? If rtl_433 can't write to MQTT, then there's no messages for the autodiscovery script to listen to either.
> 

Unfortunately, I don't know that. I didn't do anything about it. Can you please advise how to check this?

> Note we now support automatic configuration of credentials since you're using the built-in mosquitto broker. I see you migrated your configuration files to the rtl_433 directory, but are hardcoding [...]
So I have to redesign the output line in my configuration file as follows: 
`output mqtt://core-mosquitto:1883,user=homeassistant,pass=XXX,retain=${retain}`? What should I add at retain?

> And consider adding an additional `output kv` line, which will print all radio captures to the addon logs. But, I think it's likely your output line that's broken, and not something with the radio.

Unfortunately, I do not know what you mean here.

If I understand correctly, I have some outdated settings. If you please advise me how, I will be happy to redo it as it should be right now.

P-etr, 2021-12-10:
Today I reinstalled both rtl_433 add-ons and also reinstalled mqtt broker.
I reset the user for mqqt. I restarted HA.
It did not help. So I disconnected the Raspberry from the power supply for a few more minutes.
This helped and the temperature sensor at 433Mhz and the level meter at 433Mhz worked well for about an hour and I saw their data in HA via mqtt.
Now it doesn't work again and the rtl_433 log says `pulse_FSK_detect (): Maximum number of pulses reached!`

Can you please advise what to do to make my sensors work as before?

deviantintegral, 2021-12-10:
> So I have to redesign the output line in my configuration file as follows:
output mqtt://core-mosquitto:1883,user=homeassistant,pass=XXX,retain=${retain}? What should I add at retain?

Actually, you leave the above as-is, without hardcoding the host / username / password. The addon uses the supervisor API to generate access credentials and injects them automatically. I think your full configuration would be:

```
output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}
frequency   433.92M
protocol 125
protocol 165
```

> This helped and the temperature sensor at 433Mhz and the level meter at 433Mhz worked well for about an hour and I saw their data in HA via mqtt.
Now it doesn't work again and the rtl_433 log says pulse_FSK_detect (): Maximum number of pulses reached!

You may need to tweak the frequency. As the radio warms up it can drift a bit. It's the small cost of using wonderfully cheap hardware!

See https://github.com/merbanan/rtl_433/issues/1233

P-etr, 2021-12-11:
Thanks again for your answer.
Here: [](https://github.com/merbanan/rtl_433/issues/487) I read that `pulse_FSK_detect (): Maximum number of pulses reached` also means that two different sensors sent a signal at the same time on the same frequency and therefore failed to process them.
I tuned the SDR and a slightly different frequency as you advised me. This seems to have helped in part. The `pulse_FSK_detect (): Maximum number of pulses reached` error message has not yet appeared in the log since then. However, there was a loss of data from my sensors once. Both sensors failed at the same time, which seems to me that it could just be caused by the above-mentioned collision of two signals at the same time.
By analyzing the history in HA, I found that before I bought the second sensor (temperature), there was very little data loss from my first sensor (level meter) and only for a short time. Since I have two sensors, data outages are more common and sometimes longer. Tuning to a different frequency has improved, but it's still not the same as when I only had one sensor.
The new temperature sensor has a choice of three channels on which it can transmit. Do you think it makes sense to try other channels so that the signals from both sensors are not too close to each other? Or should I buy a better SDR? Maybe Nooelec?

deviantintegral, 2021-12-11:
I've only used the NESDR Mini 2+ (I have 3!), so I can't comment on other brands. Changing channels certainly seems like a good idea, but depending on how far apart the channels are you may need multiple radios anyways.

deviantintegral, 2022-01-12:
I think this has been solved as best we can from the perspective of the HA add-on. Feel free to reopen if there's a bug lurking specific to the add-on I missed.

closed 2022-01-12 by deviantintegral