pbkhrv/rtl_433-hass-addons #35: Datetime without timezone exceptions with HA 12.0

realthk, 2021-12-12:
Few minutes after upgrading to Home Assistant Core 12.0, the log is full with (hundreds) of exceptions because timestamps from RTL 433 sensors do not include timezone.
I guess this could something to do with [this change in HA](https://github.com/home-assistant/core/pull/60269)

```
2021-12-11 23:49:43 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform mqtt
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 613, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 748, in add_to_platform_finish
self.async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 505, in async_write_ha_state
self._async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _async_write_ha_state
state = self._stringify_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 511, in _stringify_state
if (state := self.state) is None:
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 411, in state
raise ValueError(
ValueError: Invalid datetime: sensor.auriol_hg02832_1_89_utc provides state '2021-12-11 23:49:23', which is missing timezone information

2021-12-11 23:49:43 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'rtl_433/9b13b3f4-rtl433/devices/Fineoffset-WH0530/36/time': '2021-12-11 23:49:20'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 37, in wrapper
msg_callback(msg)
File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 249, in message_received
self.async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 505, in async_write_ha_state
self._async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _async_write_ha_state
state = self._stringify_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 511, in _stringify_state
if (state := self.state) is None:
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 411, in state
raise ValueError(
ValueError: Invalid datetime: sensor.fineoffset_wh0530_36_utc provides state '2021-12-11 23:49:20', which is missing timezone information
```

deviantintegral, 2021-12-12:
I see this as well. I don't see an existing issue at https://github.com/merbanan/rtl_433/issues, but we should at least get a PR up fixing this and then we can patch it in the addon until a new release is made.

I'm going to file an issue now.

deviantintegral, 2021-12-12:
Filed https://github.com/merbanan/rtl_433/issues/1904

realthk, 2021-12-13:
Thanks, that workaround you've inserted above solves the problem indeed!

deviantintegral, 2021-12-14:
I appreciate you closing this, but I'm going to re-open because I'd like to get this improved upstream and then update the copy we include in the container. Spamming logs out of the box isn't a great experience for new users!

I've got some time off coming up and will hopefully get an hour or two to work on this. If anyone wants to work ahead and file a PR for https://github.com/merbanan/rtl_433/issues/1904 please feel free to mention me for reviews and testing.

deviantintegral, 2022-02-12:
This is merged and will be fixed in the next release.

closed 2022-02-12 by deviantintegral