pbkhrv/rtl_433-hass-addons #37: error:  Image 74930c0d/aarch64-addon-rtl433:0.1.2 does not exist for addon_74930c0d_rtl433

gurues, 2021-12-15:
Hello I am try install this complement but I have this error: 
Image 74930c0d/aarch64-addon-rtl433:0.1.2 does not exist for addon_74930c0d_rtl433
 
only run with RPI 32 bits?

Thanks

21494, 2021-12-29:
I have problems when start addon.
Last line o logs rtl_433:
`[s6-finish] sending all processes the TERM signal.`
In logs I have this:
`[homeassistant.components.hassio] Failed to to call /addons/9b13b3f4_rtl433/stats - 404 Client Error for http+docker://localhost/v1.41/containers/addon_9b13b3f4_rtl433/json: Not Found ("No such container: addon_9b13b3f4_rtl433")`

deviantintegral, 2021-12-29:
I believe this is the same issue as https://github.com/pbkhrv/rtl_433-hass-addons/issues/23

closed 2022-01-12 by deviantintegral