pbkhrv/rtl_433-hass-addons #38: Error installing add.on: The command '/bin/ash -o pipefail -c cmake ..' returned a non-zero code: 1

pengemetal, 2021-12-23:
Hi,
I got this error when I install the add-on
Core 2021.9.7
Supervisor: 2021.12.2
OMV 4 (Debian) docker 19.03.8

Thx

Supervisor log:
---> daa20bd29f45

Step 6/31 : WORKDIR /build

 ---> Using cache

 ---> 8493b6fa5f75

Step 7/31 : RUN git clone https://github.com/merbanan/rtl_433

 ---> Using cache

 ---> ff0bd820055e

Step 8/31 : WORKDIR ./rtl_433

 ---> Using cache

 ---> 2b0aa6bcc665

Step 9/31 : ARG rtl433GitRevision=21.12

 ---> Using cache

 ---> 158ad7e42071

Step 10/31 : RUN git checkout ${rtl433GitRevision}

 ---> Using cache

 ---> 6947275f4dea

Step 11/31 : WORKDIR ./build

 ---> Using cache

 ---> c214b12eadb8

Step 12/31 : RUN cmake ..

 ---> Running in 46c623fecaec

CMake Deprecation Warning at CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 10.3.1

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - failed

-- Check for working C compiler: /usr/bin/cc

-- Check for working C compiler: /usr/bin/cc - broken

CMake Error at /usr/share/cmake/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /build/rtl_433/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make -f Makefile cmTC_89b1a/fast && /usr/bin/make  -f CMakeFiles/cmTC_89b1a.dir/build.make CMakeFiles/cmTC_89b1a.dir/build
    make: /usr/bin/make: Operation not permitted
    make: *** [Makefile:127: cmTC_89b1a/fast] Error 127
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/build/rtl_433/build/CMakeFiles/CMakeOutput.log".
See also "/build/rtl_433/build/CMakeFiles/CMakeError.log".

Removing intermediate container 46c623fecaec

deviantintegral, 2021-12-23:
Thanks for reporting. Are you able to build the addon outside of Home Assistant? My first guess is that you have out of date HA core and Docker, so can you upgrade those both to rule them out?

pengemetal, 2021-12-23:
Thank you for your answer! Because of new Tuya integration I have avioded to upgrade HA :) I will try an outside build or take a breathe and upgrade HA.

ryan408, 2021-12-30:
I'm having the same problem here on HA OS with the most recent Core 2021.12.7,  I get the same error above.  Any other troubleshooting tips?

ryan408, 2021-12-30:
Actually, looks like mine is a different issue. I'll submit a new issue.

bennydiamond, 2022-01-10:
Currently running supervisor-2021.12.2 and homeassistant-2021.12.8 (both latest stable) and I have the same issue. 

Image does build ok outside of HASS with aarch64-base:latest as base image.

My guess is supervisor is not supplying the latest base image for building add-on images.

deviantintegral, 2022-01-12:
I believe this is the same issue as #41 , where the ARG line was _after_ `RUN git clone...`. As that issue has an open PR, I'm going to close this one. Feel free to reopen if it appears to be a separate issue. Thanks!

closed 2022-01-12 by deviantintegral

ryan408, 2022-01-12:
Thanks for looking into this one!

Closed #38<https://github.com/pbkhrv/rtl_433-hass-addons/issues/38>.