pbkhrv/rtl_433-hass-addons #39: re: rtl_433.c Hopping hack to sit on Freq1 for time and hop on success for Freq2

Ian-Zz, 2021-12-29:
LOVE THIS ADDON!!!
But I think I might need this hop after success feature seen here...
https://github.com/bobhavlin/rtl_433/commit/390b5c18d213c93c8dd9def406a0760e35d62ce3

My problem is one of my meters seems to skip an hour every now and again. 
I have set the hop rate pretty darn high and I spend half my time at a frequency/signal that has no issues and doesn't even need to be read that often. 7 or 10 seconds every 10 or 15 minutes or so would be fine for that 2nd frequency.

I am running on a pi4 with "official" HA OS install. Could I change/edit that  rtl_433.c on my local machine? After that... then what? Could I use that new parameter in the rtl_433-hass-addon?

Thanks Yall.

deviantintegral, 2021-12-29:
> But I think I might need this hop after success feature seen here...

That's neat, saving having to have multiple radios if you are recording sensors on different frequencies. Is there an upstream PR to add that to rtl_433 itself?

I don't see us changing upstreams or patching rtl_433 itself, but you're right in that you can do this locally. Use the terminal or Samba addons to clone this repo into the addons directory, and then you can edit as you see fit and rebuild the addon to use the different upstream.

https://developers.home-assistant.io/docs/add-ons/tutorial has details.

Ian-Zz, 2022-01-04:
Thanks much for the info and the link deviantintegral.
Just now getting to load my first "custom" add-ons in HA.
Fingers crossed.

Ian-Zz, 2022-01-04:
I am not 100% sure but it looks like that "hacked" code I linked to above has already been put into the original rtl_433 master.
I'm now at a loss as to how (or what) to put into my home assistant rtl conf file to get it to register that custom hop. I have not found any documentation on that setting. Not yet anyway.

Here is my HA rtl_conf.template btw...
```
output      
mqtt://my.secret.ip.bla:1883,user=user,pass=pw,retain=1
frequency   912.60M 
frequency   433.92M
protocol 8
protocol 149
protocol 154
hop_interval  800
report_meta newmodel
convert si
```

Ian-Zz, 2022-01-07:
I was wrong about the hop2 hack being in the original rtl_433 master. But I was able to create my own rtl_433 HA addon using this repo... https://github.com/bobhavlin/rtl_433. It seems to load and work well enough on my Pi4 running HA OS7 but I just cant figure out how to get the hop2 arguments to work yet through the rtl_433-hass-addons conf,template file.

As a test I loaded the https://github.com/bobhavlin/rtl_433 onto my pi3 and tested it out and it's been working great. I believe it solved my missing hours when reading my electric meter very well. This is the exact command I use to do the hop early deal... It stays on 912 for 600s and hops from 433 right after a good reading. Works perfect.
`rtl_433 -R 154 -R 8 -f 912600155 -f 433920000 -H 600 -E hop2 -F "mqtt://192.168.1.114:1883,user=user,pass=pw,retain=1,devices=rtl_600[/model][/subtype][/id][/channel]"`

My question/s now are... Is there any way to use that code snippet above that works with straight rtl_433 in a rtl_433-hass-addons .template file? and or can I add the  -E hop2 arguments to my custom/local ha rtl_433 addon myself?

deviantintegral, 2022-01-12:
For reference, the upstream issue is at https://github.com/merbanan/rtl_433/issues/1838

To build this yourself, all you should need to do is:

1. Set up access to the addons folder as mentioned above.
2. Clone this repository.
3. Change the git clone URL to that of the fork above.
4. Install the add-on.

If you do that, you'll be "stuck" at that version of the add-on and rtl_433 until you manually upgrade, or revert back from the local version.

The add-on configuration comes _after_ the add-on is built, so we don't have a ton of "easy" control over it. I think the above steps are best for testing features that are so bleeding edge that there isn't even an open upstream PR yet.

Does that work for you?

Ian-Zz, 2022-01-12:
Where I am stuck, using this add-on route is getting the hop2 to trigger like I can with my above rtl_433 start up "command". Have no clue if the hop2 or -E is usable in the add-on's conf.template as shown below. I tried a few ways but always "unknown keyword" error shows up.

`output mqtt.blah.blah
frequency   433.92M
frequency   912.60
hop_interval  500
convert     si
report_meta newmodel
# Uncomment the following line to also enable the default "table" output to the
# addon logs.
output kv`

What I have done that works for me so far. Is changed my HA OS install to a Pi4 HA Supervised install on Debian 11 and just have the hop2 version of rtl_433 run outside of HA but on the same Pi/host. Starts at boot and seems to catch every hour of energy readings = me = happy camper.

Ian-Zz, 2022-01-13:
I did use the`code thingies`. I also had a # between them. I think that fowled it up.

But anyways, I appreciate all your help again and I think for now I'm just going to stick with the Debian 11 Supervised HA install and have Bob's hop2 version of rtl_433 run outside of HA but on the same Pi/host. It works extra peachy for me.

closed 2022-01-13 by deviantintegral