pbkhrv/rtl_433-hass-addons #41: error: pathspec '21.12' did not match any file(s) known to git

ryan408, 2021-12-30:
I'm trying to re-install RTL_433 after removing it.  It randomly stopped working so was trying this as one last nuclear option since I've tried just about everything else I can think of.  I get the following in the supervisor log:

`
21-12-30 11:14:25 INFO (SyncWorker_4) [supervisor.docker.addon] Starting build for 9b13b3f4/aarch64-addon-rtl433:0.1.2
21-12-30 11:14:36 ERROR (SyncWorker_4) [supervisor.docker.addon] Can't build 9b13b3f4/aarch64-addon-rtl433:0.1.2: The command '/bin/ash -o pipefail -c git checkout ${rtl433GitRevision}' returned a non-zero code: 1
21-12-30 11:14:36 ERROR (SyncWorker_4) [supervisor.docker.addon] Build log: 
Step 1/31 : ARG BUILD_FROM
Step 2/31 : FROM $BUILD_FROM as builder
 ---> d5dbda55fa00
Step 3/31 : MAINTAINER pbkhrv@pm.me
 ---> Using cache
 ---> 3df4dba2aed0
Step 4/31 : ENV LANG C.UTF-8
 ---> Using cache
 ---> e9422cd336bb
Step 5/31 : RUN apk add --no-cache --virtual .buildDeps     build-base     libusb-dev     librtlsdr-dev     cmake     git
 ---> Using cache
 ---> eee3024f5f42
Step 6/31 : WORKDIR /build
 ---> Using cache
 ---> 6c861aecc81b
Step 7/31 : RUN git clone https://github.com/merbanan/rtl_433
 ---> Using cache
 ---> c2501db24e5a
Step 8/31 : WORKDIR ./rtl_433
 ---> Using cache
 ---> 21b6150baa56
Step 9/31 : ARG rtl433GitRevision=21.12
 ---> Using cache
 ---> e3edcfafa155
Step 10/31 : RUN git checkout ${rtl433GitRevision}
 ---> Running in d2a610b61e87
error: pathspec '21.12' did not match any file(s) known to git

Removing intermediate container d2a610b61e87
`

Any ideas?  Thanks in advance.

deviantintegral, 2021-12-30:
Interesting. So everything above says "using cache", which means at step 7 the clone is likely using an old version before 21.12 was tagged.

I think the fix is to move the `ARG` line above the `RUN git clone...` line, so any changes to it invalidate the clone cache. I'm unable to test that until (hopefully) next week, is there any chance you could try it out? If so, I'd appreciate a PR!

ryan408, 2021-12-31:
I’ve no idea where to start to modify this so I’ll have to wait for you to have a chance to look into it. Thank you. Happy New Year.

Otnow, 2022-01-10:
Applied PR #43 in my clone of the repository and the above error no longer appears.

ryan408, 2022-01-10:
Thank you for looking into this issue. I appreciate it.

PR #43 (2022-01-13): Prevent using old cached rtl_433 repository during installation or update
Fixes #41 as suggested by @deviantintegral. To recreate the issue, version v1.2 needs to be installed and then updated to v1.3.
closed 2022-02-12 by deviantintegral