pbkhrv/rtl_433-hass-addons #45: Unable to read devices after update to v0.1.2

klkohl, 2022-01-07:
This may be because I'm not configuring my USB device correctly. I'm unclear how to determine the correct USB device and the format to specify this.

This is what's in my rtl.s_433.conf.template:

# This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios, being sure to set the 'device' setting.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

device	4
frequency   915M
protocol    171
convert     customary
output      mqtt://192.168.0.147:1883,user=xxxxxxxx,pass=xxxxxxxxxxxxx,retain=0,events=rtl_433[/model][/id]

# Uncomment the following line to also enable the default "table" output to the
# addon logs.
# output kv

Here's the rtl_433 addon log:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] 
[rtl_433] New defaults active, use "-Y classic -s 250k" for the old defaults!
[rtl_433] 
[rtl_433] Publishing MQTT data to 192.168.0.147 port 1883
[rtl_433] Publishing events info to MQTT topic "rtl_433[/model][/id]".
[rtl_433] Registered 1 out of 207 device decoding protocols [ 171 ]
[cmd] /run.sh exited 2
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

Any help would be greatly appreciated.

thekitehunter, 2022-01-11:
I have the same problem, rtl_433 just stopped working at the end of December. Days of trying/reinstalling with no result, but also noticed the rtl_433.conf file kept disappearing. Managed to get it mostly working again by restoring a full backup (unfortunately from last July) and then updated to current. I believe the reason this worked was because in October I had migrated the mqtt server to a Windows server which was working fine with the origial rtl_433.conf. The restore put it all back to the defaults returning the mqtt server back onto the hassio on the raspberry pi.
Thing is, I'm not clear what is supposed to go into the new rtl_433.conf.template. Is this a substitute for the rtl433_.conf file? Are all the protocols, which used to be listed in the rtl_433.conf file now obtained by the add-on from somewhere else?
The reason I ask is that not all of my thermometers (identical make) are now automatically discovered. They all used to send messages as rtl_433/3/125/# where the '125' was unique to each unit. Having used cubicSDR to confirm that the lost unit was still transmitting a signal, I eventually found it is now messaging as 'rtl_433/9b13b3f4-rtl433/devices/Prologue-TH/5/2/135/#'.

Can you see any reason for this? Does the rtl_433.conf file need to have the original protocols added.
Thanks

deviantintegral, 2022-01-12:
> device	4

Do you have multiple radios? If not, I don't think you need to specify the device at all.

> 0.1.2

Is that a typo, or do you mean `0.1.3`? There's some missing log entries that make me think the add-on is actually running code from a ways back.

klkohl, 2022-01-12:
Thanks for responding. I think my problem is solved. I ended up removing everything from my rtl_433.conf.template file, with the exception of the frequency and output to MQTT. 

For what it's worth, I only have one sdr usb device. I am running version 0.1.3 now, but was running version 0.1.2 when I noticed the problem.

deviantintegral, 2022-01-12:
That would make sense - by forcing it to `device 4` it wouldn't even try to use `device 0`: https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf#L33-L39. Glad you got it working!

closed 2022-01-12 by deviantintegral