pbkhrv/rtl_433-hass-addons #46: protocol 154 scm+ breaks auto discovery

markwoodward86, 2022-01-07:
trying to auto discover electric meter that uses scm+ protocol. Meter is found but not added due to:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
mqtt found in this Home Assistance instance.
Starting rtl_433_mqtt_hass.py...
MQTT connected: Connection Accepted.
homeassistant/sensor/TPMS-Renault-9713f4/TPMS-Renault-9713f4-UTC/config : {"device_class": "timestamp", "name": "TPMS-Renault-9713f4-UTC", "icon": "mdi:clock-in", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/TPMS/Renault/9713f4/time", "unique_id": "TPMS-Renault-9713f4-UTC", "device": {"identifiers": "TPMS-Renault-9713f4", "name": "TPMS-Renault-9713f4", "model": "Renault", "manufacturer": "rtl_433"}, "force_update": "true"}
homeassistant/sensor/TPMS-Renault-9713f4/TPMS-Renault-9713f4-P/config : {"device_class": "pressure", "name": "TPMS-Renault-9713f4-P", "unit_of_measurement": "kPa", "value_template": "{{ value|float }}", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/TPMS/Renault/9713f4/pressure_kPa", "unique_id": "TPMS-Renault-9713f4-P", "device": {"identifiers": "TPMS-Renault-9713f4", "name": "TPMS-Renault-9713f4", "model": "Renault", "manufacturer": "rtl_433"}, "force_update": "true"}
homeassistant/sensor/TPMS-Renault-9713f4/TPMS-Renault-9713f4-T/config : {"device_class": "temperature", "name": "TPMS-Renault-9713f4-T", "unit_of_measurement": "\u00b0C", "value_template": "{{ value|float }}", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/TPMS/Renault/9713f4/temperature_C", "unique_id": "TPMS-Renault-9713f4-T", "device": {"identifiers": "TPMS-Renault-9713f4", "name": "TPMS-Renault-9713f4", "model": "Renault", "manufacturer": "rtl_433"}, "force_update": "true"}
Published TPMS/Renault/9713f4: time, pressure_kPa, temperature_C
Skipped TPMS/Renault/9713f4: flags
homeassistant/sensor/SCM+-1552820016/SCM+-1552820016-UTC/config : {"device_class": "timestamp", "name": "SCM+-1552820016-UTC", "icon": "mdi:clock-in", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/SCM+/1552820016/time", "unique_id": "SCM+-1552820016-UTC", "device": {"identifiers": "SCM+-1552820016", "name": "SCM+-1552820016", "model": "SCM+", "manufacturer": "rtl_433"}, "force_update": "true"}
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/rtl_433_mqtt_hass.py", line 515, in mqtt_message
    bridge_event_to_hass(client, topicprefix, data)
  File "/rtl_433_mqtt_hass.py", line 599, in bridge_event_to_hass
    if publish_config(mqttc, topic, model, instance, mappings[key]):
  File "/rtl_433_mqtt_hass.py", line 571, in publish_config
    mqttc.publish(path, json.dumps(config), retain=args.retain)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1233, in publish
    raise ValueError('Publish topic cannot contain wildcards.')
ValueError: Publish topic cannot contain wildcards.

deviantintegral, 2022-01-12:
> ValueError: Publish topic cannot contain wildcards.

That seems to say it all, as I think `+` is a wildcard in MQTT.

I'm not seeing [any open issues](https://github.com/merbanan/rtl_433/search?q=wildcard&type=issues) upstream for this. Can you submit there, as I don't think this is a bug in this add-on specifically.

markwoodward86, 2022-01-12:
Hi there, thank you for getting back to me on this. 

I saw the wildcard error and thought it odd that there is a protocol specifically for scm+ when it doesnt work. rtlamr does allow for this and so Ive moved to that.

You can close this one out. 

Thanks.

closed 2022-01-12 by markwoodward86