pbkhrv/rtl_433-hass-addons #54: Fineoffset-WH51 not reporting on sensors

bones3994, 2022-01-19:
Hi there. My HA crashed last year so I did a new install and copied some of my old data into the fresh install.

Strange thing is my Fineoffset-WH51 moisture % doesn't show on my sensors anymore. When I listen on MQTT I can see it is publishing, but it seems not to be picked up at the end.

Any ideas on what this might be? 

[image]

sdt16, 2022-02-09:
I wonder if this is due to #35, for me it's parsing the battery info, then blows up on the lack of TZ. I'm thinking it just doesn't even try with the moisture message.

sdt16, 2022-02-09:
Ah, no. Fixed that then got this log:

```Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'device_class': 'moisture', 'name': 'Fineoffset-WH51-00cdcc-H', 'unit_of_measurement': '%', 'value_template': '{{ value|float }}', 'state_topic': 'rtl_433/9b13b3f4-rtl433/devices/Fineoffset-WH51/00cdcc/moisture', 'unique_id': 'Fineoffset-WH51-00cdcc-H', 'device': {'identifiers': 'Fineoffset-WH51-00cdcc', 'name': 'Fineoffset-WH51-00cdcc', 'model': 'Fineoffset-WH51', 'manufacturer': 'rtl_433'}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected SensorDeviceClass for dictionary value @ data['device_class'] ```

sdt16, 2022-02-09:
I'm suspicious of this commit in HA: https://github.com/home-assistant/core/commit/12ff5dee7400516ff02dfd7c080ed756f6f802d4

sdt16, 2022-02-09:
Yep, this fixed it for me: https://github.com/sdt16/rtl_433-hass-addons/commit/727b00f611913e5caeef035acc18cd091fe1d4f2

I'll prepare a pull request with upstream at some point. 

[image]

deviantintegral, 2022-03-27:
@sdt16 I don't see a PR upstream with that, can you link if you did get it filed?

sdt16, 2022-06-08:
@deviantintegral Ack, sorry totally forgot about this. Looks like it's been fixed already: https://github.com/merbanan/rtl_433/pull/1903

deviantintegral, 2022-08-26:
rtl_433 has been updated in `next` and I expect to get a new release out soon.

closed 2023-08-04 by deviantintegral