pbkhrv/rtl_433-hass-addons #55: Invalid device class "depth"

Ulrar, 2022-01-20:
Hey,

I just installed an Apollo sensor on my kerosene tank, and it's essentially just a depth sensor to monitor how low the oil level is.
I'm not using the addon as I don't use hass, but I did copy the very handy `rtl_433_mqtt_hass.py` to have HA discover the devices.

It seems to set the device class for the depth sensor to "depth", which HA is not loving. I had to delete that line (391) to have HA accept that entity.
Looks like there's also a typo line 383 for the noise entity, it has `valuei`, I imagine someone was using vim and pressed i one too many time, I also removed that in my copy.

Thanks for the script, apart from those two details it's very useful !

deviantintegral, 2022-03-27:
@Ulrar the script actually comes from upstream, so if you're not using HAOS you can use it directly: https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py

They're pretty good about merging fixes quickly, but if you get a PR open we can always patch it in if needed. Thanks!

Ulrar, 2022-03-27:
Hi,

I had no idea, cool, thanks !
Looks like they already fixed both of these issues upstream.

closed 2022-03-27 by Ulrar