pbkhrv/rtl_433-hass-addons #57: autodiscover addon not seeing published sensor

DolfLund, 2022-01-22:
Add-on does not seem to properly recognize published sensor data.

Log says:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
mqtt found in this Home Assistance instance.
Starting rtl_433_mqtt_hass.py...
MQTT connected: Connection Accepted.

published topic looks like this:

[image]

DolfLund, 2022-01-22:
user error

closed 2022-01-22 by DolfLund

peetipablo, 2022-01-24:
@DolfLund what was your user error, if you don't mind?  I'm having the same issue where I have published sensors, but the discovery addon is not publishing discovery information.