pbkhrv/rtl_433-hass-addons #58: Protocols commented out still being detected

jayscovill, 2022-01-22:
I've commented out almost every protocol except of a few Acurite ones.  

My Acurite sensors are being detected by the SDR and add-on but so are a few other devices for protocols that I've commented out so I wonder if I'm missing something.

[image]

`#config for rtl_433

# A valid config line is a keyword followed by an argument to the end of line.
# Whitespace around the keyword is ignored, whitespace is space and tab
# Comments start with a hash sign, no inline comments, empty lines are ok.
#
# Boolean options can be true/false, yes/no, on/off, enable/disable, or 1/0
#
# All options will be applied in the order given, overwriting previous values
#
# Config files can be nested/stacked (use multiple -c and config_file = ).
#
# If no -c option is given the first found of this list will be loaded:
# - ./rtl_433.conf
# - ~/.config/rtl_433/rtl_433.conf
# - /usr/local/etc/rtl_433.conf
# - /etc/rtl_433.conf

## General options

# as command line option:
#   [-v] Increase verbosity (can be used multiple times).
#        -v : verbose, -vv : verbose decoders, -vvv : debug decoders, -vvvv : trace decoding).
# 0 = normal, 1 = verbose, 2 = verbose decoders, 3 = debug decoders, 4 = trace decoding
#verbose

# as command line option:
#   [-c <path>] Read config options from a file
#config_file

## Tuner options

# as command line option:
#   [-d <RTL-SDR USB device index>] (default: 0)
#   [-d :<RTL-SDR USB device serial (can be set with rtl_eeprom -s)>]
#   [-d "" Open default SoapySDR device
#   [-d driver=rtlsdr Open e.g. specific SoapySDR device
# default is "0" (RTL-SDR) or "" (SoapySDR)
device        0

# as command line option:
#   [-g <gain>] (default: 0 for auto)
# For RTL-SDR: gain in tenths of dB ("0" is auto).
# For SoapySDR: gain in dB for automatic distribution ("" is auto), or string of gain elements.
# E.g. "LNA=20,TIA=8,PGA=2" for LimeSDR.
gain          0

# as command line option:
#   [-t <settings>] apply a list of keyword=value settings for SoapySDR devices
# E.g. "antenna=A,bandwidth=4.5M,rfnotch_ctrl=false"
#settings      antenna=A,bandwidth=4.5M

# as command line option:
#   [-f <frequency>] [-f...] Receive frequency(s) (default: 433920000 Hz)
# default is "433.92M", other reasonable values are 315M, 345M, 915M and 868M
frequency     433.92M

# as command line option:
#   [-H <seconds>] Hop interval for polling of multiple frequencies (default: 600 seconds)
# default is "600" seconds, only used when multiple frequencies are given
hop_interval  600

# as command line option:
#   [-p <ppm_error] Correct rtl-sdr tuner frequency offset error (default: 0)
[... 415 lines cut ...]
# {"time" : "@1.572864s", "model" : "doorbell#1", "count" : 25, "num_rows" : 25, "rows" : [{"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}, {"len" : 21, "data" : "e768c8"}]}
#decoder n=Byron_BY_Doorbell,m=OOK_PWM,s=500,l=1000,r=3300,g=1200,repeats>=4,bits=21,match={2}0x3

# Kerui alarm system (PIR and door sensors)
#  short is 333 us
#  long is 972 us
#  packet gap 11000 us
#decoder n=Kerui,m=OOK_PWM,s=333,l=972,r=11000,g=1100,bits=25,invert,get={20}:state,get=@20:{4}:event:[10:pir 14:open 7:close 11:tamper 15:battery_low]

# Golden Security GS-WDS07 door and window sensor
#  short is 476 us + 1344 us
#  long is 1364 us + 448 us
#  packet gap 13972 us
#decoder n=gswds07,m=OOK_PWM,s=476,l=1364,r=15000,g=1600,bits>=24,bits<=25,invert

# Generic SCV2260 4-button remote (see rtl_433_tests/tests/generic_remote/01)
#  short is 472 us + 1412 us
#  long is 1428 us + 472 us
#decoder n=generic_remote_01,m=OOK_PWM,s=472,l=1428,r=1800,g=1600,bits=25,invert,match=13cd,get=@16:{8}:event:[192:arm 12:disarm 3:home 48:sos]

# Generic PT2260 PIR (see rtl_433_tests/tests/PT2262/01)
#  short is 440 us + 1536 us
#  long is 1428 us + 548 us
#  packet gap 15348 us
#decoder n=pt2260_pir,m=OOK_PWM,s=440,l=1428,r=16000,g=1700,bits=25,invert,match=755555,countonly`

deviantintegral, 2022-03-27:
Do you have the "retain" option in use anywhere? You may need to manually delete the MQTT topics, and the devices in HA, and then restart to see if they are still added.

deviantintegral, 2022-06-12:
Please re-open if you're still experiencing this. Thanks!

closed 2022-06-12 by deviantintegral