pbkhrv/rtl_433-hass-addons #59: Cannot use RTL stick after HassOS 7.2 upgrade

realthk, 2022-01-25:
After upgrading HassOS to 7.2, RTL433 cannot use the stick, though removing it from the USB port and re-inserting produces these kernel messages:

```
[47616.223418] usb usb3-port1: disabled by hub (EMI?), re-enabling...
[47616.223602] usb 3-1: USB disconnect, device number 2
[47616.470623] usb 3-1: new full-speed USB device number 5 using ohci-pci
[47616.566633] hassio: port 18(veth2ad7d48) entered disabled state
[47616.566886] veth65e4d11: renamed from eth0
[47616.602932] hassio: port 18(veth2ad7d48) entered disabled state
[47616.604229] device veth2ad7d48 left promiscuous mode
[47616.604258] hassio: port 18(veth2ad7d48) entered disabled state
[47616.630343] usb 3-1: not running at top speed; connect to a high speed hub
[47616.648282] usb 3-1: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
[47616.648289] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[47616.648293] usb 3-1: Product: RTL2838UHIDIR
[47616.648296] usb 3-1: Manufacturer: Realtek
[47616.648299] usb 3-1: SerialNumber: 00000001
[47680.643250] retire_capture_urb: 4 callbacks suppressed
[47725.294143] usb 3-1: USB disconnect, device number 5
[47731.247671] usb 3-1: new full-speed USB device number 6 using ohci-pci
[47731.408307] usb 3-1: not running at top speed; connect to a high speed hub
[47731.426322] usb 3-1: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
[47731.426332] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[47731.426335] usb 3-1: Product: RTL2838UHIDIR
[47731.426339] usb 3-1: Manufacturer: Realtek
[47731.426342] usb 3-1: SerialNumber: 00000001
```

and the RTL also seems to find the device (even detect its tuner) before exiting:
```
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] Publishing MQTT data to core-mosquitto port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] Registered 4 out of 207 device decoding protocols [ 16 19 79 141 ]
[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] Exact sample rate is: 250000.000414 Hz
[rtl_433] [R82XX] PLL not locked!
[rtl_433] Sample rate set to 250000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 433.920MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] Async read stalled, exiting!
[rtl_433] 
[rtl_433] LIBUSB_ERROR_NOT_FOUND: Entity not found!
[rtl_433] Check your RTL-SDR dongle, USB cables, and power supply.
[rtl_433] 
[rtl_433] WARNING: async read failed (-5).
[cmd] /run.sh exited 3
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

my conf.template is the bare minimum, only added the time format because of an earlier issues, and had no problems with HassOS 7.1
output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}

```
# Uncomment the following line to also enable the default "table" output to the
# addon logs.
# output kv

frequency   433.92M

protocol    16
protocol    19
protocol    79
protocol    141

report_meta time:iso:tz:local
```

deviantintegral, 2022-01-25:
I upgraded to 7.2 yesterday and haven't had any problems. This is inside of a VM on amd64, so perhaps this is a problem with a different architecture?

I would follow the advice and check cables and reboot to make sure it's still a problem. If it is, this should be reported upstream as we don't do anything with the kernel or devices in the addon.

realthk, 2022-01-25:
No VM: it is running on a HP T620 thin client directly, and upgrading to HassOS 7.2 of course included a reboot. RTL 433 was working fine before, last data from the sensors came right before the reboot and nothing else was changed.
I'll try a full cold start with power-off tonight.

realthk, 2022-01-25:
Well, after a complete power off it finally works again!
Sorry for the false alarm, it has gone wrong at the upgrade and it's weird that according to these log messages, both OS and RTL 433 detected the device, yet it wasn't working (and pulling out from the USB also was not enough) so I thought perhaps the addon might use something that is missing from 7.2.

closed 2022-01-25 by realthk

realthk, 2022-03-22:
Weird, the same has happened after upgrading HassOS to 7.5, with the very same "async read failed" error, even though the RTL stick and its tuner was detected... 
A simple reboot solved it, but strange, anyway.