pbkhrv/rtl_433-hass-addons #62: TPMS sensors that include temp data being published

imagesafari, 2022-02-05:
I'm getting lots of TPMS sensors that include temperature data being published. A change, or an option to ignore all ^TPMS* sensors would be nice.

mcavarano, 2022-03-10:
Yah my mqtt is up to like 2500 devices because I live by a pretty busy road

deviantintegral, 2022-03-13:
In the short term, the best thing to do is to only use the autodiscovery addon when trying to add new devices. This was added as step 5 in the readme: https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433_mqtt_autodiscovery#installation

Long term, if someone submits a PR to https://github.com/merbanan/rtl_433 adding a feature for this in https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py I'd certainly consider adding it in here even before it's merged.

deviantintegral, 2022-10-19:
Since commenting on this, I learned that you can not only allow specific protocols, but you can disable them as well. For new installs, the latest release will start with the configuration disabling known TPMS sensors. Check the release notes for the latest release notes for copy-pasteable lines to put into existing configurations.

closed 2022-10-19 by deviantintegral