pbkhrv/rtl_433-hass-addons #72: light_lux not in supported keys list, only lux

joanwa, 2022-02-19:
I came across `rtl_433 MQTT Auto Discovery` logging

`Skipped Cotech-367959/111: light_lux` - where `light_lux` is not in the supported keys list stated in the README. A complete json payload from my switchdoclabs weatherstation is
```
[rtl_433] rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
[rtl_433] {"time" : "2022-02-19T22:40:20+0100", "model" : "Cotech-367959", "id" : 111, "battery_ok" : 0, "temperature_C" : 5.667, "humidity" : 60, "rain_mm" : 39.500, "wind_dir_deg" : 116, "wind_avg_m_s" : 2.200, "wind_max_m_s" : 3.600, "light_lux" : 0, "uv" : 0, "mic" : "CRC"}
```
Not sure if the supported keys follow some specific conventions, but it looks like `measuredthing_unit`. The README says `lux` is a supported key, but `lux` would only be a unit without the thing itself, and `light_lux` would follow the same as `rain_mm` or `temperature_C`.

So I would suggest to add `light_lux` as a supported key in the same manner as `lux`.

deviantintegral, 2022-02-19:
I just finished tagging a new release, so upgrading now should fix this. Be sure to see the notes in the CHANGELOG. Thanks!

joanwa, 2022-02-20:
just fyi, also found this:
https://github.com/merbanan/rtl_433/issues/1986

joanwa, 2022-02-21:
Shall I make a PR or do you want to do it? The upstream fix above is already merged to master, so it can be pulled in again

deviantintegral, 2022-02-24:
Ideally all that needs to be changed is https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/Dockerfile#L16, but there may be conflicts in the patch to resolve. Feel free to get something started, as I likely won't get to it for at least a few days.

deviantintegral, 2022-02-24:
Oh, and https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433_mqtt_autodiscovery/Dockerfile#L4 too.

deviantintegral, 2022-06-12:
This should be fixed as of the previous `0.3.3` addon release.

closed 2022-06-12 by deviantintegral