pbkhrv/rtl_433-hass-addons #73: install error "manifest unknown" for v0.2.0

joanwa, 2022-02-20:
trying to uninstall and re-install as recommended in the CHANGELOG. Getting:

```
Failed to install add-on
Can't install ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433-aarch64:0.2.0: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=0.2.0&fromImage=ghcr.io%2Fpbkhrv%2Frtl_433-hass-addons-rtl_433-aarch64: Internal Server Error ("manifest unknown")
```

ellaizee, 2022-02-20:
I'm experiencing the same issue.

deviantintegral, 2022-02-20:
Yeah, unfortunately I set up my RPi 4 when 32-bit was recommended, and I missed adding `aarch64` to the build list. This was fixed in https://github.com/pbkhrv/rtl_433-hass-addons/pull/74 so the latest release should do it. You may need to trigger "Check updates..." in the addon store if it's not showing up. Feel free to reopen if for some reason 0.2.1 isn't working.

closed 2022-02-20 by deviantintegral