pbkhrv/rtl_433-hass-addons #76: Allow autodiscover to be used with standalone docker

ScottRoach, 2022-02-20:
I run Home Assistant through docker, so add-ons are not available. I was hoping to be able to run this same container in a standalone docker environment.

It looks like [run.sh](https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433_mqtt_autodiscovery/run.sh#L11-L12) finds arguments to the python file from bashio. Is there an equivalent way to pass arguments just through docker?

I tried running roughly the following:
```
docker run --rm -e MQTT_HOST=mqtt.com -e MQTT_PORT=1883 -e MQTT_USERNAME=username -e MQTT_PASSWORD=password --name mqtt_auto ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64
```
And get the following:
```
curl: (6) Could not resolve host: supervisor
[07:19:03] ERROR: Something went wrong contacting the API
Starting rtl_433_mqtt_hass.py...
usage: rtl_433_mqtt_hass.py [-h] [-d] [-q] [-u USER] [-P PASSWORD] [-H HOST]
                            [-p PORT] [-r] [-f] [-R RTL_TOPIC]
                            [-D DISCOVERY_PREFIX] [-i DISCOVERY_INTERVAL]
rtl_433_mqtt_hass.py: error: argument -p/--port: invalid int value: 'null'
```

deviantintegral, 2022-02-20:
We could extend `run.sh` to only call into `bashio` if the variable isn't defined.

My first thought was "why not just use the upstream images", but I see https://github.com/hertzg/rtl_433_docker doesn't actually have a variant with Python included. Given this would help with certain types of testing, I'd be open to a PR for this.

closed 2022-06-12 by deviantintegral

dlasher, 2024-01-11:
Seeing the same issue, using current version - how do you pass the IP address/hostname for "supervisor"?

`curl: (6) Could not resolve host: supervisor`

dlasher, 2024-01-11:
ok, worked around that.

`    extra_hosts:
      - "supervisor:10.10.10.10"
`

However, my supervisor isn't running on port 80, it's a non-standard port. How do you add supervisor port to the config?