pbkhrv/rtl_433-hass-addons #78: Autostart issue and issues with frequency

stomko11, 2022-02-23:
Hello. I have few issues. I am using latest version of Supervised install of HA. Despite I disabled autostart, addon still starts with HA boot.

In addition, I am having weird issues with my config. To eliminate other issues, I keep my config minimal for now, with only one line:
```
output      mqtt://a.b.c.d:1883,user=xxxx,pass=yyyy
```
However when I start the addon, I get the following:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
/tmp/rtl_433_heredoc: line 2: warning: here-document at line 1 delimited by end-of-file (wanted `EOD')
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 21.12-51-g2b49b81a branch  at 202201311039 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] Publishing MQTT data to a.b.c.d port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] Registered 181 out of 212 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-212 ]
[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] Exact sample rate is: 250000.000414 Hz
[rtl_433] [R82XX] PLL not locked!
[rtl_433] Sample rate set to 250000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 433.920MHz.
[rtl_433] Allocating 15 zero-copy buffers
[rtl_433] MQTT Connected...
[rtl_433] baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
[rtl_433] MQTT Connection failed...
[rtl_433] MQTT Connected...
[rtl_433] MQTT Connection error: 5
```
This goes on forever. My password or username don't contain any special characters.

when I stop the addon and specify frequency in addition:
```
frequency   433.92M
```
It does not start at all and hangs at:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
/tmp/rtl_433_heredoc: line 3: warning: here-document at line 1 delimited by end-of-file (wanted `EOD')
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 21.12-51-g2b49b81a branch  at 202201311039 inputs file rtl_tcp RTL-SDR
```

As soon as I change that frequency to 868M, it works immediately, without any MQTT issues:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
/tmp/rtl_433_heredoc: line 3: warning: here-document at line 1 delimited by end-of-file (wanted `EOD')
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version 21.12-51-g2b49b81a branch  at 202201311039 inputs file rtl_tcp RTL-SDR
[rtl_433] Use -h for usage help and see https://triq.org/ for documentation.
[rtl_433] Publishing MQTT data to a.b.c.d port 1883
[rtl_433] Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433] Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433] Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433] 
[rtl_433] New defaults active, use "-Y classic -s 250k" for the old defaults!
[rtl_433] 
[rtl_433] Registered 181 out of 212 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-212 ]
[rtl_433] Found Rafael Micro R820T tuner
[rtl_433] Exact sample rate is: 1000000.026491 Hz
[rtl_433] [R82XX] PLL not locked!
[rtl_433] Sample rate set to 1000000 S/s.
[rtl_433] Tuner gain set to Auto.
[rtl_433] Tuned to 868.000MHz.
[rtl_433] Allocating 15 zero-copy buffers
```

If I restart HA (core), keep the config file empty, and start the addon, it again hangs at:
```
[rtl_433] rtl_433 version 21.12-51-g2b49b81a branch  at 202201311039 inputs file rtl_tcp RTL-SDR
```

Basically I am unable to get 433,92MHz to work. Either by some miracle, I am able to start it, but get MQTT related errors, or I don't at all. If I change frequency to 868MHz, without any HA/Core restarts, everything works well.

Any ideas?

deviantintegral, 2022-02-24:
That's quite confusing! Can you test with https://github.com/hertzg/rtl_433_docker using the `hertzg/rtl_433:21.12` tag and `hertzg/rtl_433:master`?

I'm pretty sure error 5 means "not authorized" in MQTT. You could also try adding `output kv` to log to the console too.

stomko11, 2022-02-24:
Can you advise how to try it? I am using HA addon, not docker.

stomko11, 2022-02-27:
OK I did try the same using my Raspberry Pi. First I followed `apt-get` install method and tried using `rtl_433` without any further configuration, and it worked just fine using default frequency 433.92 MHz.

```
pi@raspberrypi:~ $ sudo rtl_433
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-02-27 23:26:47                    brand     : OS
model     : Oregon-THGR122N                        House Code: 29
Channel   : 1            Battery   : 0             Temperature: -0.50 C      Humidity  : 35 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-02-27 23:26:47                    brand     : OS
model     : Oregon-THGR122N                        House Code: 29
Channel   : 1            Battery   : 0             Temperature: -0.50 C      Humidity  : 35 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-02-27 23:26:48
model     : AmbientWeather-TX8300                  id        : 40
channel   : 1            Battery   : 2             Temperature: -2.3 C       Humidity  : 52 %          MIC       : CHECKSUM

```

Then I also tried the docker using first master branch:
```
pi@raspberrypi:~ $ sudo docker run --device /dev/bus/usb/001/006 hertzg/rtl_433:master
Unable to find image 'hertzg/rtl_433:master' locally
master: Pulling from hertzg/rtl_433
5480d2ca1740: Pull complete
bcf5fcd54c18: Pull complete
4dcbf6d18321: Pull complete
Digest: sha256:42a2252ace5d7804cf19a7946307e856c7c39d18324bfb8acf12264f107073a8
Status: Downloaded newer image for hertzg/rtl_433:master
rtl_433 version 21.12 (2021-12-14) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 182 out of 213 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-213 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

And then also `debian-21.12` branch:
```
pi@raspberrypi:~ $ sudo docker run --device /dev/bus/usb/001/006 hertzg/rtl_433:debian-21.12
Unable to find image 'hertzg/rtl_433:debian-21.12' locally
debian-21.12: Pulling from hertzg/rtl_433
046501ac4c58: Pull complete
035cce8225bb: Pull complete
36d04fbfb12d: Pull complete
[... 28 lines cut ...]
Battery   : 0
Temperature: -0.70 C
Humidity  : 38 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : 2022-02-27 22:54:05
model     : Oregon-THGR122N
House Code: 29
Channel   : 1
Battery   : 0
Temperature: -0.70 C
Humidity  : 38 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : 2022-02-27 22:54:05
model     : Oregon-THGR122N
House Code: 29
Channel   : 1
Battery   : 0
Temperature: -0.70 C
Humidity  : 38 %

```

Note that I don't own the sensor yet, so I did not always wait for actual sensor data, just whether it starts correctly or not.

stomko11, 2022-03-08:
For now, I am running rtl_433 in standalone docker outside of HA.
I have downloaded and installed addon just now, but it still uses (I assume) old image, as there are already new protocol definitions available in rtl_433, but addon itself contains only 212. How do I make sure addon downloads the latest image available?

deviantintegral, 2022-03-13:
Thanks for all of the testing above. To test on your actual Home Assistant install, you can follow these steps to enable SSH into the OS host (if you don't have a keyboard and monitor), and when you're logged in you can run all of the usual docker commands like you've already done: https://developers.home-assistant.io/docs/operating-system/debugging/

When you mentioned "raspberry pi" - is that the same hardware your Home Assistant install is on? If not, I suppose it's possible there's some kind of USB hardware issue.

The image is only built when we create a new release here. The most recent addon version is `0.2.1`, which currently builds from this commit on master (current as of early February): https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433/Dockerfile#L16

What I'd like to do eventually is compile both the latest tag and current master release in, and add an option to switch between them: https://github.com/pbkhrv/rtl_433-hass-addons/issues/49

stomko11, 2022-03-14:
I have ssh enabled on my HA instance (i run VM), but it does not seem like I can run 'docker run' commands or 'rtl_433 ....`.
When I was testing on rpi, this was completely different Raspberry pi, just to prove that my tuner is OK, and running it with docker image actually works.
For now, I use rtl_433 as a standalone docker as I 1) can't get it to run in HA 2) protocols added lately are not included in the plugin

deviantintegral, 2022-03-16:
If you're in a VM, you can also just connect to the VM console too.

Are you using the terminal and SSH addon? If so, that doesn't have access to docker itself and would explain things. You have to use the above instructions, and connect on port `22222`.

deviantintegral, 2022-08-26:
Please reopen if you're able to confirm this is an issue with the latest release. Thanks!

closed 2022-08-26 by deviantintegral