pbkhrv/rtl_433-hass-addons #79: Using MQTT formats prevents rtl_433 from starting

nukefrenzy, 2022-02-24:
When I just use the following everything runs fine:

`output mqtt://192.168.1.13:1883,user=mqtt,pass=MYPASSWORD`

However, as soon as I start trying to change the topic structure, the addon just hangs and won't continue. Here's an example of what I'm trying to do:

`output mqtt://192.168.1.13:1883,user=mqtt,pass=MYPASSWORD,devices=rtl_433[/id]`

I have confirmed with the rtl_433 devs that my output string is correct and works with their code.  See here: https://github.com/merbanan/rtl_433/issues/1995

The below is all I see in the logs:

[image]

deviantintegral, 2022-03-27:
I just gave this a try, and it looks to be working for me. Here's my whole configuration file:

```
device 0
output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain},devices=asdf[/id]
report_meta time:iso:tz:local
convert si
output kv
verbose
```

[image]

nukefrenzy, 2022-04-30:
I pasted what you have above and it doesn't seem to work.  What do your logs show when using that config file?

nukefrenzy, 2022-06-07:
Finally got some time to play around with this more.  I installed the MQTT Explorer app and I'm able to see the messages as they come in and everything is working fine with the "devices" info added to the output line.  What is weird is that the log just stops as shown in my original post whenever I add "devices".  When I remove it I get the full log output as shown below.  That's what tripped me up, I thought the add-on hung up.  I wonder why adding an extra parameter to the output would change the logging?

[image]

deviantintegral, 2022-06-08:
We have a bug with output buffering that can cause logs to be delayed until there’s further input.

are you putting device before or after the output line? I think it needs to be first so it exists to reference in the output line.

nukefrenzy, 2022-06-09:
Update: putting 'device 0' first does not fix the logging issue.  My logs still look like the screenshot in my first post.

deviantintegral, 2022-10-19:
I've tagged a new release today which fixes the output buffering. Are you still having any trouble with this?

nukefrenzy, 2022-10-20:
Everything looks normal in my logs again and everything is working well. Thanks for the update!

closed 2022-10-20 by nukefrenzy