pbkhrv/rtl_433-hass-addons #80: Device added many times

hw-monkey, 2022-03-12:
I only have 3 devices at the moment. For two, it works perfectly.  For the other, it keeps adding new devices (although I see other devices that are not mine that have this same problem). This makes it impossible to use for the entity is continually changing.  It creates 254 devices for the one physical device.

Instead of 34, other devices have a different sequential number. 

A topic is
`homeassistant/sensor/Cotech-367959-34/Cotech-367959-34-H/config`
Payload is 

```
device_class: humidity
name: Cotech-367959-34-H
unit_of_measurement: '%'
value_template: '{{ value|float }}'
state_class: measurement
state_topic: rtl_433/9b13b3f4-rtl433/devices/Cotech-367959/34/humidity
unique_id: Cotech-367959-34-H
device:
  identifiers:
    - Cotech-367959-34
  name: Cotech-367959-34
  model: Cotech-367959
  manufacturer: rtl_433
platform: mqtt

Subscribed topics:
rtl_433/9b13b3f4-rtl433/devices/Cotech-367959/34/humidity
```

How can I get these to act like a single device and not be discovered 254 times?

deviantintegral, 2022-03-13:
According to https://github.com/merbanan/rtl_433/blob/5f0ff6db624270a4598958ab9dd79bb385ced3ef/src/devices/cotech_36_7959.c#L29 the ID should only change when the sensor is reset. Is it possible you have a weak battery, or a reset button is stuck down?

If you can, please add `output kv` to your configuration. If the ID is changing there, then there isn't much we can do that wouldn't negatively affect other users.

hw-monkey, 2022-03-14:
The battery is weak, but the device is still constantly on.

I am unclear about your request. Please clarify. I believe you want me to edit the rtl_433 MQTT Auto Discovery yaml to include:
`output: kv`
Is that what you mean?
And then what? Should one of the logs show something different?

deviantintegral, 2022-03-16:
Add that to your `rtl_433.conf.template` file after the existing `output` line. No colon! That will tell rtl_433 to both print out what it reads _and_ send them over mqtt.

GeorgeIoak, 2022-07-13:
Just a note that the same has happened to me several times. I just put new batteries in the 2 sensors and have deleted all the extra devices in HA (last week was over 100, today was 77).

GeorgeIoak, 2022-11-15:
Just a follow up to say that this continues to happen to me. I'm confused why this is happening and how I can track down what is causing it. I just cleared out all the extra/abandoned sensors about a week a ago and I'm seeing new sensors again. I rename the front and back yard sensors (front is xxx-2-20 and back is 1-33) and now I am seeing 1-2 and 5-33. Normally when I need to clean up extra sensors I'll know that they are bogus because they don't show a battery percentage. 

Just looking for some clues on how I might go about getting to the bottom of this

[image]

deviantintegral, 2022-11-17:
1. Have you turned off the autodiscovery addon from automatically starting? If you aren't intentionally trying to add a new device, only running it when needed is best. See step 5 at https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433_mqtt_autodiscovery#installation
2. I was testing some changes in an unrelated device upstream in rtl_433, and I noticed that the ID detected for my rain meter changed on older rtl_433 releases. This was due to a bug in the decoder that's now fixed. It's possible something similar is happening to you?

GeorgeIoak, 2022-11-17:
No, I had been running it and missed that step 5 says that it should be run only as needed which makes sense. I've kept rtl_433 up to date (through HA) so if that was a fixed bug it still happens to me as I just found new devices again when I went to turn off. It still would be nice to figure out why this happens though out of curiosity but at least this is off now so my problem should be fixed.

deviantintegral, 2024-02-17:
Sounds like this is fixed, so closing. Thanks!

closed 2024-02-17 by deviantintegral