pbkhrv/rtl_433-hass-addons #84: Error - Exception in message_received when handling msg

twinkie0101, 2022-04-05:
I got rtl_433 up and running today and was able to add my two acurite sensors, but the following error keeps showing up in my logs and not really sure if 1) should I be concerned and 2) what do I need to do to correct the error. The devices are updating.

```
Logger: homeassistant.util.logging
Source: util/logging.py:114
First occurred: 10:53:05 PM (3 occurrences)
Last logged: 10:53:05 PM

Exception in message_received when handling msg on 'rtl_433/9b13b3f4-rtl433/devices/Acurite-986/1R/53349/time': '2022-04-04 22:53:04' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 47, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 284, in message_received self.async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 553, in async_write_ha_state self._async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 590, in _async_write_ha_state state = self._stringify_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 559, in _stringify_state if (state := self.state) is None: File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 382, in state raise ValueError( ValueError: Invalid datetime: sensor.fridge_temperature_sensor_utc provides state '2022-04-04 22:53:04', which is missing timezone information
```
My config is simple as I am only adding the acurite sensors at the moment;
```

# This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios, being sure to set the 'device' setting.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

output      mqtt://core-mosquitto:1883,user=user_id,pass=password
protocol    41
convert     si
report_meta newmodel

# Uncomment the following line to also enable the default "table" output to the
# addon logs.
output kv
```
If any additional information would be helpful, happy to provide, just let me know.

deviantintegral, 2022-04-05:
There's nothing really to be concerned about here. It means that the "last updated" sensor in HA won't be updated. I have `report_meta time:iso:tz:local` in my weatherstation config to reformat the time to include the time zone.

I also reported this uptream at https://github.com/merbanan/rtl_433/issues/1904.

twinkie0101, 2022-04-05:
Thanks for the quick response! I am going to close this issue.

closed 2022-04-05 by twinkie0101