pbkhrv/rtl_433-hass-addons #86: rtl433 battery status is not properly handled

Gonioul, 2022-04-20:
Shows as 100% without icon
Wheareas other integrations shows battery icon

Gonioul, 2022-04-20:
[image]

Gonioul, 2022-04-20:
[image]

Gonioul, 2022-04-20:
[image]

Gonioul, 2022-04-24:
Looks like it should be an integer

deviantintegral, 2022-04-25:
Thanks for the report. Are you thinking the problem is with the `value_template` line at https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py#L162-L173 ?

Gonioul, 2022-04-26:
I have no idea... But converting the level to an integer seems logical if it's a 0-100 range.

Gonioul, 2022-05-03:
Another difference, the same sensor with rfxcom (top) and rtl_433 (below).
It didn't rename the rtl_433 entities for now, so it's the default name.

[image]

[image]

Gonioul, 2022-05-03:
EnCh - Entity Checker for AppDaemon doesn't seem to recognize the low battery level with rtl_433, works with rfxtrx, dunno why, no trace in logs. Maybe the float value again.

deviantintegral, 2022-05-09:
I think it would be good if we could confirm if Home Assistant is expecting ints or floats for battery percentages. I haven't found docs or code yet confirming one way or another.

jaseelye, 2022-05-24:
This is what I use and it works great.
"value_template": "{% if is_number(value) %}{% set temp_val = float(value) * 100 | float(default=0) %}{{ temp_val | float(default=0) | round(0) }}{% endif %}"

deviantintegral, 2024-02-17:
Since this is from the upstream code, and I don't have any devices that expose this battery info to test, I'm closing this out. Please feel free to open a new issue if a PR is filed upstream we should patch in to improve this. Thanks!

closed 2024-02-17 by deviantintegral