pbkhrv/rtl_433-hass-addons #88: No Data after Core Update or Supervisor retart

LaCruz75, 2022-04-21:
There are no data from rtl_433, when a core update was performed or supervisor was  restartet. Only reboot host helps.

LaCruz75, 2022-04-28:
ok, I turned logging in for rtl_433, there seems data come in. So the core restart issue belongs to the addon "rtl_433 MQTT Auto Discovery"
 After manually restart this addon after a core restart, data come in. 
So, how to fix that?

deviantintegral, 2022-04-28:
Do you have mqtt_retain turned on in the auto discovery addon? If not, that would explain it.

LaCruz75, 2022-05-02:
havn't set mqtt_retain.. 
If that is need, why is standard not true?
I had no idea if I needed that.

Thanx for your help.

deviantintegral, 2022-05-09:
I'm actually not sure why it defaults to `false`.  It was added over at https://github.com/pbkhrv/rtl_433-hass-addons/commit/41b6e03bbcb1acf1fb66d7915637042e829ba669 but that doesn't explain the thinking behind it.

I'm not sure it's completely "required". For example, I think with a device that sends updates often it will get reregistered quickly. If it's longer, such as a door sensor that only updates every few hours unless it's opened, it may lag.

What kind of sensor were you trying to read, and to confirm flipping that setting worked?

deviantintegral, 2022-06-12:
Let's change the default at https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433_mqtt_autodiscovery/config.json#L16 from `false` to `true` to finish up this issue.