pbkhrv/rtl_433-hass-addons #89: when will standalone support be available on published docker image?

andre-pt, 2022-04-30:
hi

I would like to use the standalone feature which was merged (https://github.com/pbkhrv/rtl_433-hass-addons/commit/60347e1c50d41c368174c702f275a9aa89e55ebd) but it's not part of the published docker images.

The publish step is skipped on the build job: https://github.com/pbkhrv/rtl_433-hass-addons/runs/5802459064?check_suite_focus=true

is this intended? when will it be available?

thanks a lot

deviantintegral, 2022-05-09:
This has been available since the last release. For some reason, GitHub isn't showing the containers like normal in the sidebar. There must be a repo setting I don't have access to. You can pull directly with `ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64:0.3.2` and `ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433-amd64:0.2.2` (changing the arch as needed).

The build logs are at https://github.com/pbkhrv/rtl_433-hass-addons/actions/runs/2083337053.

For this PR, a README update with the image names and tags would be a reasonable next step. Contributions welcome!

@pbkhrv offered to set up a GitHub org for this, and at the time I thought it was added complexity and would be a pain to migrate users over. Given that I've been able to (mostly) maintain this for a while now, and that I now understand we could keep this repo with nothing but `config.json` or as a mirror, perhaps we revisit that.

andre-pt, 2022-05-14:
I think there is some misunderstanding, sorry for it.

if i pull the latest or 0.3.2 images, the run.sh is not the latest.

```
$ docker run --rm -it ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64 /bin/sh
```
then inside of the container:
```
$ cat /run.sh
#!/usr/bin/env bashio

if bashio::services.available mqtt; then
  echo "mqtt found in this Home Assistance instance."
  MQTT_HOST=$(bashio::services mqtt "host")
  MQTT_PORT=$(bashio::services mqtt "port")
  export MQTT_USERNAME=$(bashio::services mqtt "username")
  export MQTT_PASSWORD=$(bashio::services mqtt "password")
else
  echo "Using an external mqtt broker."
  MQTT_HOST=$(bashio::config "mqtt_host")
  MQTT_PORT=$(bashio::config "mqtt_port")
  export MQTT_USERNAME=$(bashio::config "mqtt_user")
  export MQTT_PASSWORD=$(bashio::config "mqtt_password")
fi

RTL_TOPIC=$(bashio::config "rtl_topic")
DISCOVERY_PREFIX=$(bashio::config "discovery_prefix")
DISCOVERY_INTERVAL=$(bashio::config "discovery_interval")

OTHER_ARGS=""
if bashio::config.true "mqtt_retain"; then
  OTHER_ARGS="${OTHER_ARGS} --retain"
fi
if bashio::config.true "force_update"; then
  OTHER_ARGS="${OTHER_ARGS} --force_update"
fi

echo "Starting rtl_433_mqtt_hass.py..."
python3 -u /rtl_433_mqtt_hass.py -d -H $MQTT_HOST -p $MQTT_PORT -R "$RTL_TOPIC" -D "$DISCOVERY_PREFIX" -i $DISCOVERY_INTERVAL $OTHER_ARGS
```

This is not the same file as https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433_mqtt_autodiscovery/run.sh
I guess there was some issue publishing this latest docker image?

Regards
Andre

deviantintegral, 2022-05-16:
Ah! `latest` is not currently configured to be pushed with releases. It floats for open PRs. Can you specify a specific tag in your command?

`docker run --rm ghcr.io/pbkhrv/rtl_433-hass-addons-rtl_433_mqtt_autodiscovery-amd64:0.3.3 cat /run.sh` should show the updated run.sh.

andre-pt, 2022-05-29:
ah my bad, i was always pulling `latest`
thanks

closed 2022-05-29 by andre-pt