pbkhrv/rtl_433-hass-addons #90: rtl_433_mqtt_hass.py Hectopascals present but not Kilopascals....

randysaz928, 2022-05-02:
Sorry I am new and dont know how to properly do a pull request or for that matter does anyone care about the "examples".
The rtl_433_mqtt.hass.py script is missing the below and quire common pressure definition for Kilopascals... 

    "pressure_kPa": {
        "device_type": "sensor",
        "object_suffix": "P",
        "config": {
            "device_class": "pressure",
            "name": "Pressure",
            "unit_of_measurement": "kPa",
            "value_template": "{{ value|float }}"
        }
    },

deviantintegral, 2022-05-09:
Interesting, I see it added 12 months ago at https://github.com/merbanan/rtl_433/blame/master/examples/rtl_433_mqtt_hass.py#L211

I just double checked the `0.3.2` release and it's there. Can you confirm you're running the latest version of the addon?

deviantintegral, 2022-06-12:
Feel free to reopen if you're still having trouble after upgrading to the latest release.

closed 2022-06-12 by deviantintegral