pbkhrv/rtl_433-hass-addons #91: Rolling Code change

Gonioul, 2022-05-05:
Hi,

Most 433 weather sensors change Rolling Code when batteries are changed.

With rtl_433 and rtl_433_mqtt_autodiscovery, a new sensor is created and there seems to be no way to replace the old sensor and keep history.
The RFXTRX integration also creates a new sensor but is able to replace the old one and so keep history.

I know it's probably not a rtl_433 problem, maybe a MQTT or Home Assistant problem, but a quick doc would be nice as I found nothing on the subject.

Regards,

deviantintegral, 2022-05-09:
I'm curious about "most". Last time I changed batteries on my AccuWeather station, I didn't have to do any sort of pairing. But perhaps lower end devices don't save the current ID to nvram?

Actually, I don't think weather stations have rolling codes. If they did, you wouldn't be able to passively read them without pairing somehow.

What kind of weather sensor do you have?

deviantintegral, 2022-05-10:
Ok, so you're seeing the channel or ID change? The keys used are at https://github.com/merbanan/rtl_433/blob/9eec461132ee880c4e1a969026f81be9934682cf/examples/rtl_433_mqtt_hass.py#L103-L104

Unfortunately, this is likely not possible to solve. Unless there's a message that maps an old ID to a new one, I don't see how we can know that a sensor is actually the same.

As well, Home Assistant is doing the "right" thing by keeping history attached to the same entity, even if the ID changes. One possible workaround is to add influxdb and grafana to your stack. I use them myself, and you can preserve history since influx just stores a list of data points over time.

deviantintegral, 2022-06-12:
I wonder if you can work around this by manually changing the `sensor` topic in the Home Assistant MQTT config. For example, mine has `homeassistant/sensor/LaCrosse-TX141THBv2-0-166/LaCrosse-TX141THBv2-0-166-T/config` with:

```json
{
  "device_class": "temperature",
  "name": "LaCrosse-TX141THBv2-0-166-T",
  "unit_of_measurement": "°C",
  "value_template": "{{ value|float }}",
  "state_class": "measurement",
  "state_topic": "rtl_433/local-rtl433/devices/LaCrosse-TX141THBv2/0/166/temperature_C",
  "unique_id": "LaCrosse-TX141THBv2-0-166-T",
  "device": {
    "identifiers": [
      "LaCrosse-TX141THBv2-0-166"
    ],
    "name": "LaCrosse-TX141THBv2-0-166",
    "model": "LaCrosse-TX141THBv2",
    "manufacturer": "rtl_433"
  }
}
```

Perhaps you can update the `unique_id` to match the old one? Regardless, I'm closing this as the best place to solve this is upstream in the rtl_433 project, if it's even possible. Thanks for reporting.

closed 2022-06-12 by deviantintegral