pbkhrv/rtl_433-hass-addons #93: Last-updated

Gonioul, 2022-05-19:
Hi,

Looks like the last-updated value of the entity isn't updated if the sensor value doesn't change (temperature for example).

The only way to know the entity is not dead is to look for the utc value of the device.

Regards,

deviantintegral, 2022-05-25:
Good thought! My sensors all update fairly frequently with different values. Can you check to see if rtl_433 is sending the value to mqtt?

My first guess is we may want to set `force_update`, which defaults to false. If that's the case, we should change this upstream first at https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py.

deviantintegral, 2022-06-12:
> I confirmed on the Home Assistant Discord that setting force_update: true would fix this, but it's unclear to me if / how this is supported using this Add On with the Auto Discovery Add On too.

It is! There is a `force_update` setting in the autodiscovery configuration:
[image]

Does that work for you? If so, I think we need to mention it in the readme as it's not there currently.

deviantintegral, 2022-08-26:
There's open discussions about this at https://github.com/merbanan/rtl_433/pull/2102 and https://github.com/home-assistant/home-assistant.io/pull/23900.

rct, 2022-08-28:
Looking into this, I still need to get a better understanding of updated, changed, availability…

first comment, it seems like force update for all 433 auto discovered devices is a pretty blunt control.  Does this wind up generating state updates in hass / recorder every time a message is received?

deviantintegral, 2022-09-19:
> first comment, it seems like force update for all 433 auto discovered devices is a pretty blunt control. Does this wind up generating state updates in hass / recorder every time a message is received?

Yes, that's exactly the issue. It's [suggested here](https://github.com/home-assistant/home-assistant.io/pull/23900#discussion_r969348745) that removing `force_update` entirely may be a better solution.

@Gonioul , it turns out that "Last updated" doesn't actually mean what it looks like in the UI. I've filed a discussion in the Home Assistant front-end repository at https://github.com/home-assistant/frontend/discussions/13687.

To solve the original case of "The only way to know the entity is not dead is to look for the utc value of the device.", I think we'd need to write something that marks an entity as unavailable if a given period of time has passed.

deviantintegral, 2022-09-19:
Ah! We can expose expire-after perhaps: https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py#L747

kohkohwastaken, 2023-01-15:
Registering my interest in this!