pbkhrv/rtl_433-hass-addons #103: usb_claim_interface error -6

stomko11, 2022-07-30:
I am running addon on Supervised install on my Intel NUC, OS 8.4.
I am getting following error when starting the addon:
```
usb_claim_interface error -6
```
I have been using this SDR dongle on Raspberry Pi before without any issues. Can't seem to find a way to use it on HA. Any help would be appreciated.

deviantintegral, 2022-08-26:
Is there anything else running that could possibly be using the device? https://github.com/hertzg/rtl_433_docker/discussions/34

I see others mentioning needing to blacklist the DVB module. However, if you're running HAOS that shouldn't be required.

Chartreusito, 2023-06-03:
@stomko11 were you able to pinpoint the issue ? I'm having the same problem on HAOS. What bothers me is that the dongle worked fine for a few hours at first

stomko11, 2023-06-03:
@Chartreusito no, I switched to docker version running on unraid and did not use HA addon

deviantintegral, 2023-06-09:
I've seen this myself since this was posted. I've actually had trouble with both of my NESDR Mini 2+ radios where they drop until I reset them or reboot the server. I even switched out rtl_433 to run on the host and not use USB passthrough to the HAOS VM, and then connect to to the addon via rtl_tcp to debug.

_however_ I also have a NESDR Mini 2+ on a Raspberry Pi, and it is rock solid. So, perhaps there's some subtle compatibility issue with the server's motherboard or the USB hub I'm using.

Anyways, I guess my point is that using rtl_tcp is a great way to debug, because you can run it on any other system without making significant changes on the HA side. Hope that helps.

Chartreusito, 2023-06-09:
Well, i persisted to try it on HAOS, and what i realized is that i had left the configuration template "rtl_433.conf.template" active. Once i removed i had no more `usb_claim_interface error -6`, but i can't say i'm happy with how it works, i'll try on a dedicated vm.

deviantintegral, 2024-02-17:
I don't see an action here for the addon (especially since this is all HAOS / hardware related). So, I'm closing for now, but feel free to open a new issue if there's something this layer of the stack can improve on here.

closed 2024-02-17 by deviantintegral