pbkhrv/rtl_433-hass-addons #105: Cant see scm meter any longer

morbidpete, 2022-08-19:
Last time I received data from my meter was 08-15-2022,
       running rtlamr2mqtt shows the meter data fine. But I need to run rtl_433 to also read my sensors. unsure why it broke. conf file is pretty simple and has not changed

```
output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain}
frequency   433.92M
frequency   915M
protocol    154
protocol    149
protocol    40
protocol    176
convert     si
report_meta newmodel
hop_interval  60
```
any thoughts? 
[image]

deviantintegral, 2022-08-21:
To reflect back:

1. You have a single NESDR radio.
2. When running rtl_433, you are using frequency hopping to get data from different sensors.
3. Your home assistant automation swaps between rtl_433 and rtlamr every 30 minutes.
4. Everything is working fine with rtl_433, but rtlamr no longer gets data.

In that case, I doubt this addon in particular is causing problems. I suppose it's possible all of the tuning has got it to a point where the USB radio has bugged out. Did you try a reboot? Another possibility is that your scm data interval has somehow changed, and you're missing it in the 30 minute window.

Another thought is to pull the radio to another computer and verify you can get rtlamr data. Finally, it may just be easier to buy a second radio, avoiding the need for the automation at all.

morbidpete, 2022-08-21:
> To reflect back:
>
> 1. You have a single NESDR radio.
> 2. When running rtl_433, you are using frequency hopping to get data from different sensors.
> 3. Your home assistant automation swaps between [...]

1. correct
2. Also correct. Has been working for almost a year. All of a sidden jus the power meter with scm is an issue.
3. The automation is a (hopefully) temp solution to turn on rtlamr to get me meter reading (reports every 10 seconds) then go back to rtl_433 for the accurate sensor data
4. rtl_433 gets the accurate data with protocol 40 no problem. SCM (154 or 149) doesn’t work now. rtlamr gets the scm data without issue. I could grab another radio I guess but would like to see if I can get it resolved this way first.

deviantintegral, 2022-08-26:
@morbidpete if you can, I would run your radio directly with https://github.com/hertzg/rtl_433_docker (or rtl_433 itself) outside of your home assistant instance, against `master`. That way we can know if it's an issue with rtl_433 or with our packaging and configuration of it.

morbidpete, 2022-08-26:
> @morbidpete if you can, I would run your radio directly with https://github.com/hertzg/rtl_433_docker (or rtl_433 itself) outside of your home assistant instance, against `master`. That way we can know if it's an issue with rtl_433 or with our packaging and configuration of it.

Can do. I'll throw it on my docker host this weekend and see how it reacts.

dimatx, 2022-09-18:
@morbidpete - curious if you have made any progress on this. I have a similar setup where rtlamr read my water meter (R900) without a hitch, but I am unable to get it to read with rtl_433. I use an automation in HA to swap between running each of the addons to get my water meter reading once a day. Would love to consolidate and rely on rtl_433 exclusively.

deviantintegral, 2024-02-17:
Closing as there's been no recent activity. Feel free to reopen if there's something new here.

closed 2024-02-17 by deviantintegral