pbkhrv/rtl_433-hass-addons #110: Removing auto-discovered devices

eblot, 2022-09-04:
Hi, 

Is there a way to remove RTL433 devices that have been autodiscovered?
The plugin has generated >200 entries, mostly from random TPMS sensors from the nearby streets, but even removing the autodiscovery plugin does not help, as the devices are associated w/ the MQTT plugin, not its auto-discovery companion chip.

I do not want to remove MQTT plugin since it tracks many other device...

So is there a way to get rid of all those one-time, unwanted device entries from my HA? Thanks.

beerygaz, 2022-09-05:
Was just about to install this to help autodiscover some sensors when I thought the same thing - how do I filter, or specifically exclude some devices like TPMS and other remote controls.  I guess some sort of topic filter in a yaml file? 
```
---
exclude:
  - /TPMS/Shrader/0AXXYY 

```
would work, but how to regularly edit that file I'm not sure, as I don't know if plugins have access to the HA /config directory.

deviantintegral, 2022-09-19:
We've changed the install instructions to not recommend running the autodiscovery addon all of the time, and only when trying to discover new devices. But, that doesn't completely solve the problem. For example, perhaps a sensor only transmits once every hour and many cars drive within range in that time.

It looks like rtl_433 supports _disabling_ protocols now.

> Specify a negative number to disable a device decoding protocol (can be used multiple times)

I'm thinking we could disable these protocols by default in the initial configuration file. We'd need to watch for new TPMS protocols being added, but that can just be a step when updating the underlying `rtl_433` version. Thoughts?

```
rtl_433 -R help 2>&1 | grep -i tpms
    [59]  Steelmate TPMS
    [60]  Schrader TPMS
    [82]  Citroen TPMS
    [88]  Toyota TPMS
    [89]  Ford TPMS
    [90]  Renault TPMS
    [95]  Schrader TPMS EG53MA4, PA66GF35
    [110]  PMV-107J (Toyota) TPMS
    [123]* Jansite TPMS Model TY02S
    [140]  Elantra2012 TPMS
    [156]  Abarth 124 Spider TPMS
    [168]  Schrader TPMS SMD3MA4 (Subaru)
    [180]  Jansite TPMS Model Solar
    [186]  Hyundai TPMS (VDO)
    [201]  Unbranded SolarTPMS for trucks
    [203]  Porsche Boxster/Cayman TPMS
```

> Is there a way to remove RTL433 devices that have been autodiscovered?

I think you can do this by removing the topics under `homeassistant/<type>/...`. I do think you'd have to manually delete the devices as well in Home Assistant. I checked and I don't see a way in https://github.com/home-assistant-ecosystem/home-assistant-cli to do a bulk delete, but perhaps that could be written.

deviantintegral, 2022-10-19:
New installs disable known TPMS sensors by default. We'll need to add notes to the changelog when we see new sensor protocols being added.

closed 2022-10-19 by deviantintegral

eblot, 2022-12-25:
FWIW, I finally found a way to recover a "clean" installation without those hundreds of parasitic entries with the help of SQLite -to clean up data- and JQ -to clean up the configuration- and make those entries disappear from the UI:

Data are stored into a hidden directory (why?!):

Assuming a supervised installation on a Debian host:

Stop HA, then

```shell
cd /usr/share/hassio/homeassistant

## SQLite
sudo cp home-assistant_v2.db home-assistant_v2.db.backup
sudo sqlite3 home-assistant_v2.db

# dump parasitic entries
sqlite> select statistic_id from statistics_meta;
# delete each invalid device kind, ex. here with TPMS
sqlite> delete from statistics_meta where statistic_id like 'sensor.tpms%';
sqlite> vacuum;
# exit SQLite interpreter
sqlite> .quit

## JQ
jq <.storage/core.device_registry 'del(."data" | ."devices"|.[]| select(.manufacturer == "rtl_433"))' > /tmp/core.device_registry
sudo mv /tmp/core.device_registry .storage/core.device_registry
```

Restart HA

YMMV. Do not forget to backup your HA installation first...

PR #113 (2022-10-19): Disable TPMS sensors by default #110
## Summary

This PR updates the template to disable TPMS sensors by default. The list of protocols is from `master` in rtl_433.

Fixes #110

## Alternatives Considered

We could add some sort of exclusion from the autodiscovery script, but this will both reduce the amount of MQTT traffic and doesn't require writing any new code.

## Testing Steps

1. Move any existing configuration files out of the way.
2. Start or restart the rtl_433 addon.
3. TPMS sensors should not be logged.