pbkhrv/rtl_433-hass-addons #111: Device Not Found | Raspberry Pi 4 64-bit Home Assistant Operating System workaround found

msholly, 2022-09-26:
Hello! First, I wanted to say thanks for all of the work on this addon. It's one of my favorite parts of my Home Assistant install. 

I recently swapped around my RPi, and started running HA on my Raspberry Pi 4 from the previous RPi 3 Model B. My Nooelec NESDR Mini USB would not appear and the RTL433 Addon would exit. Re-installing both of your addons didn't make a difference, along with some other troubleshooting I tried. 

I did get Home Assistant and your addon running on the Raspberry Pi 4 32-bit image, however. So no real action item for you, but I wanted to report my issue in case anyone else had trouble with the RPi 4 64-bit. 

Thanks so much!

deviantintegral, 2022-09-27:
I also have a Pi 4, but it's running the 32-bit image as well. I've been meaning to switch it over at some point. If you feel like troubleshooting further, you could enable SSH with https://developers.home-assistant.io/docs/operating-system/debugging/, and then use `lsusb` and `dmseg` to make sure the NESDR Mini shows up, and that there aren't any USB-related error messages.

msholly, 2022-09-27:
I still have a separate SD card with 64-bit HA, I'll swap it out and run those commands and post back here!

thegreekgeek, 2023-01-08:
I also have this issue, but I'm just trying to run the autodiscovery container in docker. I'm not well versed in building containers, so I can't rebuild it for my system (linux/arm64/v8). Any plans to support arm64 architectures?

deviantintegral, 2023-01-09:
I only have one Pi, so it’s tricky for me to support both architectures. It’s also at a remote location, and the HA install is shared with other family members.

Are you able to run the above steps to get at least basic USB info from from lsusb and dmesg?

One challenge is for all we know, many others are using the 64-bit builds without issue. I might put a PSA into the next release asking for feedback here.

deviantintegral, 2024-02-17:
I've since switched that install to 64-bit HAOS, and it's been fine for many months. I think this was either not an issue or was fixed in HASO since.

closed 2024-02-17 by deviantintegral