pbkhrv/rtl_433-hass-addons #114: Potential breaking change in HA core update 2022.10.4?

daviswieck, 2022-10-18:
### The problem

My add on has been running perfectly fine for the past 1-2 years. Just recently it is no longer sending messages via mqtt. 
The logs of the addon say it is connected to MQTT but I cannot see any messages it is "sending". 

### What addon are you reporting the bug for?

rtl_44

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[08:26:40] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
Starting rtl_433 -c /config/rtl_433/433.conf.template
rtl_433 version 21.12-101-g9eec4611 branch  at 202204282249 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Publishing MQTT data to core-mosquitto port 1883
Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
newmodel option (-M) is deprecated.
Registered 19 out of 218 device decoding protocols [ 11 30 40-41 48 84 97 112-113 130 136 142 160-163 190 192-193 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
MQTT Connected...
MQTT Connection established.

Supervisor Logs
Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:123
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: October 17, 2022 at 1:04:02 PM (11 occurrences)
Last logged: 8:20:27 AM

Failed to to call /addons/9b13b3f4_rtl433/stats - Container addon_9b13b3f4_rtl433 is not running
Failed to to call /addons/9b13b3f4_rtl433/stats - Can't read stats from addon_9b13b3f4_rtl433: Expecting value: line 1 column 1 (char 0)
```

### Additional information

I am running HA OS via Virtual Machine. Not sure what is going on all of the sudden. Not sure if the "core-2022.10.4" update broke it or not.

deviantintegral, 2022-10-19:
I'm also running in a VM, and haven't had any issues with 2022.10.4.

> Failed to to call /addons/9b13b3f4_rtl433/stats - Container addon_9b13b3f4_rtl433 is not running

This indicates the addon is exiting. Can you add `verbose` and `output kv` to your config?

What version of the addon to you have?

daviswieck, 2022-10-19:
RTL_433 is 0.2.3

Addon Logs
```[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[08:13:11] WARNING: rtl_433 now supports automatic configuration and multiple radios. The rtl_433_conf_file option is deprecated. See the documentation for migration instructions.
rtl_433 version 21.12-101-g9eec4611 branch  at 202204282249 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Publishing MQTT data to core-mosquitto port 1883
Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
Registered 19 out of 218 device decoding protocols [ 11 30 40-41 48 84 97 112-113 130 136 142 160-163 190 192-193 ]
Starting rtl_433 -c /config/rtl_433/433.conf.template
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
MQTT Connected...
MQTT Connection established.
```

Supervisor Logs
```22-10-19 08:13:09 INFO (SyncWorker_1) [supervisor.docker.interface] Stopping addon_9b13b3f4_rtl433 application
22-10-19 08:13:10 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_9b13b3f4_rtl433 application
```

deviantintegral, 2022-10-19:
I just tagged a new release with several fixes including the end-of-file error. Mind updating and trying again?

daviswieck, 2022-10-19:
I've also opened it up to catch all signals and I don't see anything coming in.

daviswieck, 2022-10-19:
Appears it may have something to do with this:

https://github.com/home-assistant/supervisor/issues/3919

deviantintegral, 2022-10-19:
Ok, great. I'll close this but feel free to open if you think there is something in this addon to fix. Thanks!

closed 2022-10-19 by deviantintegral