pbkhrv/rtl_433-hass-addons #117: rtl_433 upstream fix for too many digits

joanwa, 2022-10-20:
### The problem

just got a rtl_433 PR merged to fix the too many displayed digits https://github.com/merbanan/rtl_433/pull/2209

which fixes home assistant values like

4.77778 °C for temperature
or
227.10001 mm for the rain sensor

Perhaps it's worth doing a micro version bump and get that upstream in?

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.3.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

deviantintegral, 2022-10-20:
I've been working on https://github.com/merbanan/rtl_433/pull/2102 so I'll hopefully get a new build and release out in the next few days.

In the meantime, we can also get a build on `next` done by bumping the git revisions in each Dockerfile. If you file a PR, that will get containers built with the tag `pr#-merge` and you can pull and override the image tag on your local.

Also, since this is the discovery script, I think this is actually for the discovery addon and not the rtl_433 addon.

deviantintegral, 2022-10-25:
Merged! I'll be getting a new release out within the next day.

closed 2022-10-25 by deviantintegral

joanwa, 2022-10-25:
Great, and sorry that I haven't been responsive