pbkhrv/rtl_433-hass-addons #121: Invalid values from rtl_433

joanwa, 2022-10-25:
### The problem

Every second day or so I observe values coming from rtl_433 that are out of bounds, mostly multiple readings at the same time. Reporting it here, but probably needs to be fixed in rtl_433 value templates or if someone has an idea on where best to fix this.

[image]
[image]
[image]

### What addon are you reporting the bug for?

rtl_443

### What is the addon version?

0.3.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
see attached pictures
```

deviantintegral, 2022-10-25:
Yes, as long as you can confirm via `output kv` or looking in the MQTT messages, we'll know that this is an `rtl_433` issue.

I ran into something similar with my rain meter. I wonder if the decoder you're using is having trouble with corrupt messages too. See https://github.com/merbanan/rtl_433/issues/2105 for investigation hints.

deviantintegral, 2024-02-17:
I'm closing this because the addon just takes rtl_433 values as is. If there's a patch upstream that improves this that we should consider patching in, feel free to open a new issue.

closed 2024-02-17 by deviantintegral