pbkhrv/rtl_433-hass-addons #123: CurrentCost-TX Power values not parsed

p6sfrx725z28knfy, 2022-11-07:
### The problem

The CurrentCost Energy monitor devices are supported by RTL_433, but as they have 3 power clamp inputs the results are exported in a different format from what it seems the rtl_433-hass-addon is expecting..

Here is the output from rtl_433 itself:

time      : 2022-11-07 16:09:34
model     : CurrentCost-TX                         Device Id : 2033
Power 0   : 1750 W       Power 1   : 0 W           Power 2   : 0 W

I note this is not the only energy monitor with more than one power value, for example an OWL CM180i:

time      : 2022-11-07 16:13:26
model     : Oregon-CM180i House Code: 6608
Battery   : 0            Power1    : 112 W         Power2    : 32216 W       Power3    : 46593 W       sequence number: 1

I see somebody else reported this in #101 which I will attempt later, but it would be nice if these devices would work "out of the box" since they are supported by rtl_433 :)

### What addon are you reporting the bug for?

rtl_433_mqtt_autodiscover

### What is the addon version?

0.5.0

### What type of MQTT Broker are you using?

Home Assistant Mosquitto MQTT Broker

### Addon log messages

```txt
mqtt found in this Home Assistance instance.
Starting rtl_433_mqtt_hass.py...
INFO:root:Discovering all devices
INFO:root:MQTT connected: Connection Accepted.
INFO:root:Subscribing to: rtl_433/+/events
INFO:root:Published CurrentCost-TX/2033: time
INFO:root:Skipped CurrentCost-TX/2033: power0_W, power1_W, power2_W
```

deviantintegral, 2022-11-08:
This addon just wraps the upstream discovery script into a HA-compatible docker container. If someone files a PR upstream at https://github.com/merbanan/rtl_433 and it's looking good, I'd consider patching it in the build process before it's merged.

deviantintegral, 2024-02-17:
If this gets added upstream, feel free to open a new issue with details. Thanks!

closed 2024-02-17 by deviantintegral